lumapu/ahoy #741: Offset für Produktion und suche nach mqtt-Cloud Anbieter

Parodontitis, 2023-03-06:
Auf meine Empfehlung hat mein Nachbar ein Balkonkraftwerk mit einem gebrauchten HM-600 Wechselrichter gekauft. Dass man den Counter nicht zurücksetzen kann habe ich bereits gelesen. Meine Fragen:

1. Gibt es ein **Offset** den ich setzen könnte, damit mein Nachbar seine Produktion ab "0Wh" sehen könnte? 
2. Da bei meinem Nachbar 0 Computer-Affinität nachgewiesen wurde, will er das "ersparte Geld" doch noch irgendwie sehen. Welchen mqtt-Cloud Anbieter könnt ihr mir empfehlen? Die Anforderungen sind einfach. 1 Device, kostenlos, schöne bunte Diagramme, Übersicht über Tag/Woche/Monat, genug Datenempfang für 1-5 Minuten Frequenz.

Vielen Dank.

stefan123t, 2023-03-06:
Offset gibt es in den Settings. Aber einen einfachen Cloud Anbieter können wir Dir nicht empfehlen, da wir ja gerade ohne einen solchen auskommen wollen, deswegen machen wir uns ja die Mühe. Wenn Du einfach den Meßwert willst, dann hänge / klemme doch einfach ein Meßgerät dazwischen, die kommen meist mit einem Cloudangebot in EU und kosten ca. 20,- Euro. Die bieten zwar idR nur 1 Stunde Auflösung aber bei 0 Computer-Affinität braucht man vermutlich auch keine genauere Statistik um sich am Ende des Jahres über den gewonnenen / gesparten Strom zu freuen.

Loetnase, 2023-03-06:
Schaut euch mal die Produkte von Shelly an, entweder das Shelly Plus 1PM zum Einbau oder Shelly PLUS Plug S zum dazwischen stecken, beide kosten Richtung 20-25€, es gibt aber noch weitere bessere und teurere ...EM. 
Shelly hat da eine kostenlose Cloud welche doch einiges anzeigt. Das Shelly Plus 1PM habe ich auch im Einsatz. 
Du benötigst nur jemanden der solch ein Shelly fachgerecht einbaut, es ist aber nicht aufwendig. WLAN muss aber vorhanden sein. 
Die Shellys sind aber auch nicht rücksetzbar soviel ich weiß, neu fangen die an bei 0Wh zu zählen.

So sieht das gerade an diesem trüben Tag aus, aktuell kommen 70,2 W rein und heute wurden 64,04 Wh erzeugt.
Mit Download werden die stündlichen Werte in Excel exportiert.
[image]
Das war der Februar mit 44,23 kWh
[image]
Und dieses Jahr habe ich bereits 68,31 kWh erzeugt
[image]

Thomas-SW, 2023-03-10:
Auch wenn es nicht ganz zur Frage paßt:
für 0 Computeraffinität wäre wohl ein einphasiger Zähler, der lokal abgelesen werden kann, echt besser.
Spart die Dauerbetreuung und jede Menge Nerven. ;-)
Kostet als digitale Version so um die 15 - 20 € und kommt ohne irgendeinen www-Schnickschnack aus.
Wird einfach in die Zuleitung eingeschleift und zählt dann drauf los, was eingespeist wird.
[nur ein Beispiel, soll keine Werbung sein](https://www.ebay.de/itm/155009356488)
Achtung: die einfachen Dinger zählen immer "PLUS", egal, wie die Leistung fließt...

closed 2023-03-25 by lumapu