lumapu/ahoy #743: [ESP8266] Gradzeichen fehlt bei Temperaturanzeige

QuickStopGroceries, 2023-03-06:
### Platform

ESP8266

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.93

### Github Hash

90955b9

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

In der Liveansicht fehlt bei der Temperaturangabe das Gradzeichen "°" vor dem "C".

Und: Gibt's eigentlich eine Möglichkeit, die Sprache auf Deutsch umzustellen?

lumapu, 2023-03-07:
danke für den Hinweis. Bisher gibt's noch keine Deutsche Sprache, bin aber schon am überlegen wie das eingebunden werden kann.

mtet2, 2023-03-07:
Und bei Gelegenheit noch in der "TOTAL" Ansicht **var** gegen **VAr** ersetzen.

knickohr, 2023-03-07:
Der Gradkringel ist jetzt in der DEV 94 drin und auch das VAr im grünen Bereich. Im orangen/braunen ist es noch var 😉

roku133, 2023-03-07:
Dann ist es wenigstens noch an einer Stelle korrekt 😉, nämlich im orange/braunen Bereich. Das Einheitenzeichen zur Blindleistung ist var und nicht VAr  - siehe ausführliche Diskussion in #732  ...

lumapu, 2023-03-07:
ihr macht mich fertig

closed 2023-03-27 by lumapu