lumapu/ahoy #744: Web Scrape einer entfernten live-Webseite

Mart-Muc, 2023-03-07:
Hallo Zusammen,
ich habe mehrere Hoymiles +Ahoy an verschiedeneren Standorten, teils ohne smarthome (HA).
Da mqtt zusammenschalten somit nicht in Frage kommt (und auch zu kompliziert), will ich die Werte direkt aus der Webseite nehmen.
(analog Simon42)

HA/Scrape bekomm ich nicht zum laufen.... kann mir ein HA-Spezialist mit dem selector helfen? Als Ergebnis kommt nur "Unbekannt"

Bzw Frage @lumapu : können eindeutige ids für die Felder vergeben werden? (anstatt 4. child)

configuration.yaml

scrape:
  - resource: http://x.x.x.x/live/
    sensor:
      - name: "Hoymiles P AC"
        unique_id: 'hoymiles_p_ac'
        select: "#live > div > div.ch-iv > div:nth-child(4) > span.value"
        # unit_of_measurement: "W"
        # device_class: "power"

PS: falsch ich im falschen Forum bin, sorry

lumapu, 2023-03-07:
am besten die JSON API verwenden, da gibt es auch alle Werte

`[IP]/api/live`
`[IP]/api/inverter/id/0` - `[IP]/api/inverter/id/9`

Mart-Muc, 2023-03-08:
Habe es hiermit geschafft:

Als Tip für die community:
https://community.home-assistant.io/t/how-to-extract-from-nested-json/99768/18

  - platform: rest
    resource: http://192.168.169.8/api/live
    name: Rothenburg PV Leistung
    unique_id: "rothenburg_pv_leistung"
    value_template: "{{ value_json.inverter[0].ch[0][2] |int }}"
    unit_of_measurement: "W"  

Danke Lukas :-)
Wo kann ich dir einen Kaffee kaufen / Spende fürs Projekt?

closed 2023-03-08 by Mart-Muc

lumapu, 2023-03-08:
sehr gerne 😉, ich habe aber auch die Änderung gemacht, die deine bisherige Auswertung unbrauchbar gemacht hat. 🙈

Bei Github sind auf der rechten Seite zwei Möglichkeiten zu spenden, vielen Dank 😊