lumapu/ahoy #746: development version 0.5.95 Nokia-Display fehlt plötzlich in den Settings

nuiler, 2023-03-08:
### Platform

ESP8266

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

development version 0.5.95

### Github Hash

f5ef19c

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Seit dem Update auf development version 0.5.95 fehlt das Nokia-Display. Ist dies gewollt?

knickohr, 2023-03-08:
Wurde wegen Stabilitätsproblemen und Verwendung anderer Ports vom ESP8266 entfernt. Wird nur noch am ESP32 (HSPI, andere Pins) supportet.

nuiler, 2023-03-08:
Schade - gab es einen Hinweis darauf?

Loetnase, 2023-03-08:
@knickohr 
Das find ich auch schade.
Ich habe da so schöne Platinchen (PCB) für den ESP8266 fertigen lasse und auch schon was mit dem Nokia aufgebaut aber die Oleds funktionieren auch, ich bin fast soweit um die hier für alle online zu stellen. Siehe  #733
Ist es denkbar einen ESP32 D1mini mit gleicher Pinbelegung wie ESP8266 zu verwenden.
Wären da alle PIOs entsprechend ummappbar.

knickohr, 2023-03-08:
Dazu habe es mehrere Diskussionen imDiscord, auch gab es im Github Issues. Daraufhin wurde beschlossen das Nikia vorerst beim ESP8266 mal nicht über den SPI gleichzeitig mit dem NRF-Modul kommunizieren zu lassen. Der ESP32 hat 3 getrennt SPI, deshalb haben wir eine strikte Trennung eingeführt.

lumapu, 2023-03-08:
@Loetnase 
Habe gestern noch mit @dAjaY85 diskutiert, es kann gut sein, dass das Nokia in kürze beim ESP8266 wieder reinkommt.

lumapu, 2023-03-08:
sollte wieder gehen mit Version `0.5.96`

Loetnase, 2023-03-09:
Danke euch allen. 
Ihr seid toll.

dAjaY85, 2023-03-09:
Geht das Nokia wieder, hat es jemand getestet?

Loetnase, 2023-03-09:
Bei mir hat es leider nicht funktioniert.
Das Update von 0.5.93 auf 0.5.96 ergab einen blanken Bildschirm und nicht mehr bootenden ESP 8266.

knickohr, 2023-03-09:
Ich habe es auch nicht hin bekommen. Sieht auch so aus als ob Settings nicht übernommen werden. Beispiel OLED und Drehung, Kontrast (Helligkeit) ohne Funktion.

lumapu, 2023-03-09:
OLED und Drehung konnte ich erfolgreich anwenden, aber wie ihr mich kennt ist die `0.5.97` schon bald unterwegs und behebt hoffentlich alle Probleme mit Displays

Loetnase, 2023-03-10:
@lumapu 
bei mir funktionier es in der 0.5.97 (rechts schwarzer Hintergrund) leider nicht wie gedacht.
Das Nokia Display zeigt wieder etwas an aber nur Offline weil die  **nrf24l01+ is not connected**
Bei der 0.5.93 (links weisser Hintergrund) funktionier es, da geht die Hardware online.
[image]
[image]
Rotation funktioniert nicht, pixel shifting funktioniert
[image]
Hast du ein Nokia zum Testen, wenn nein könnte ich dir eines komplett aufgebaut am Wemos zukommen lassen, 
als kleines Dankeschön für deine Anstrengungen für uns alle. 
[image]

dAjaY85, 2023-03-11:
Moin, ich habe ein Nokia.

Mit dieser Hardwareverdrahtung soll es auf der dev 97 funktionieren.

[image]

closed 2023-03-27 by lumapu