lumapu/ahoy #747: development version 0.5.95 - keine Kommunikation mit HM-800 und sonstiges

tqma1, 2023-03-08:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.95

### Github Hash

f5ef19c

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Nach dem Update von 0.5.93 auf 0.5.95 keine Kommunikation mehr mit meinem HM800 mehr möglich.
Downgrade auf 0.5.93 - alles wie gehabt

Ebenso aufgefallen ist mir das nach dem Update alle Display Settings weg waren und sich das Gerät nicht automatisch mehr mit meinem WLAN verbunden hat.

lumapu, 2023-03-08:
kann ich so nicht bestätigen, bei mir läuft alles wie vorher, HM1200 antwortet, sie Daten sind da.
WLAN müsste ich nicht neu eingeben.

Beim Display wurde was verändert, das ist bekannt. Hier müssen evtl die Einstellungen erneut eingegeben werden.

tqma1, 2023-03-08:
Läuft bei Dir am ESP8266 oder ESP32?
Werde nochmals testen und berichten.

niekniek89, 2023-03-08:
No issues here either after updating to 0.5.95.
My HM-1500 with 3 panels are seen and read.
Use an ESP8266.

lumapu, 2023-03-08:
bin auch auf dem ESP8266

tqma1, 2023-03-08:
Nachfolgendes Ergebnis:
Sobald ich die Display Einstellungen vornehme ist keine Kommunikation mit dem Wechselrichter mehr möglich.
Ohne Display Konfiguration funktioniert alles einwandfrei.
Es reicht schon wenn ich nur die Display Type wähle (SH1106 1,3).
Wenn ich dann die PIN Belegung mache Clock = D1 GPIO5 SCL und Data = D2 GPIO4 SDA bringt das keine Änderung

MiniOh, 2023-03-08:
Bei mir ist auch die WLan Konfig weg, aber das hast du ja oben beschrieben.
Kann ich das WLan per Console neu setzen?

Bei mir funktioniert das Speichern der WLAN Settings per GUI seit einigen Versionen eher schlecht als recht.
Ich muss oft zw. 5 und 15 mal die WLan Settings eintragen und Speichern, bis diese dann auch nach dem Neustart übernommen werden.
Warum das so ist, habe ich noch nicht wirklich herausfinden können.

tqma1, 2023-03-08:
ESP32 + SH1106 funktioniert einwandfrei.
@MiniOh Probleme mit den WLAN-Settings mit ESP32 oder ESP8266
versuche mal das WLAN-Passwort nicht einzutippen sondern einzufügen ....

MiniOh, 2023-03-08:
ESP32
OK, werde ich versuchen.

MiniOh, 2023-03-08:
Keinen Unterschied:

Console gibt folgendes aus:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
I: Initializing FS ..
I:  .. done
I: Settings valid: true
I: RF24 Amp Pwr: RF24_PA_I: LOW
I: Radio Config:
SPI Frequency		= 1 Mhz
Packets lost on
Retry attempts made for

---------
AhoyDTU Info:
Version: 0.5.95
Github Hash: f5ef19c

---------
AP MODE
SSID: AHOY-DTU
PWD: esp_8266
IP Address: http://192.168.4.1
---------

Wifi event: 0
Wifi event: 2
Wifi event: 10
Wifi event: 12
connect to network 'WLAN' ...
I: added inverter HMxxxx
I: added inverter HMxxxx
I: added inverter HMxxxx
I: added inverter HMxxxx
I: added inverter HMxxxx
I: added inverter HMxxxx
I: added inverter HMxxxx
I: added inverter HMxxxx
I: resetPayload: id: 0
I: resetPayload: id: 1
I: resetPayload: id: 2
I: resetPayload: id: 3
I: resetPayload: id: 4
I: resetPayload: id: 5
I: resetPayload: id: 6
I: resetPayload: id: 7
I: resetPayload: id: 8
I: resetPayload: id: 9
I: resetPayload: id: 0
I: resetPayload: id: 1
I: resetPayload: id: 2
I: resetPayload: id: 3
I: resetPayload: id: 4
I: resetPayload: id: 5
I: resetPayload: id: 6
I: resetPayload: id: 7
I: resetPayload: id: 8
I: resetPayload: id: 9
E (1369) i2c: i2c_set_pin(873): scl and sda gpio numbers are the same
[  1375][E][esp32-hal-i2c.c:88] i2cInit(): i2c_param_config failed
[  1677][E][esp32-hal-i2c.c:260] i2cSetClock(): bus is not initialized
[  1677][E][Wire.cpp:526] write(): NULL TX buffer pointer
[  1677][E][Wire.cpp:526] write(): NULL TX buffer pointer
[  1682][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer

lumapu, 2023-03-08:
das stimmt was an den Settings nicht `scl and sda gpio numbers are the same`

MiniOh, 2023-03-08:
Ich hatte weder an den Settings noch am Pinout was verändert, lediglich das Update von 0.5.90 auf 95 gemacht.

lumapu, 2023-03-08:
ja gerade bei der neuesten Version hat sich am Pinout, bzw. dessen Speicherung was verändert, das kann gut für Ursache für die Warnung sein.

lumapu, 2023-03-09:
verstehe nicht, warum GPIO0 vorausgewählt ist, meine Erwartung war off. Dem muss ich noch nachgehen

MiniOh, 2023-03-09:
Hier sind auch alle Display Pinouts auf GPIO0 vorausgewählt.

lumapu, 2023-03-09:
dann ist es wohl so programmiert, ich prüfe es nochmal

Argafal, 2023-03-27:
Tritt dieses Problem mit Release 0.6.0 noch auf?

closed 2023-04-05 by lumapu