lumapu/ahoy #748: [ESP8622] Build not working

deeplymbedded, 2023-03-08:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.66
Github Hash: f8fe044

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [x] Platform IO

### Desktop OS:
* [x] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [x] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Build is not successfull for all ESP86x targets. This is the console output for ESP8622-release:

Processing esp8266-release (platform: espressif8266; board: esp12e; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: gf8fe044
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2022.12.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
[... 588 lines cut ...]
                 from utils/dbg.h:30,
                 from app.h:10,
                 from app.cpp:11:
/home/kja2li/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:71:18: note: String::String(double, unsigned char)
         explicit String(double, unsigned char decimalPlaces = 2);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:70:18: note: String::String(float, unsigned char)
         explicit String(float, unsigned char decimalPlaces = 2);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:69:18: note: String::String(long unsigned int, unsigned char)
         explicit String(unsigned long, unsigned char base = 10);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:68:18: note: String::String(long int, unsigned char)
         explicit String(long, unsigned char base = 10);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:67:18: note: String::String(unsigned int, unsigned char)
         explicit String(unsigned int, unsigned char base = 10);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:66:18: note: String::String(int, unsigned char)
         explicit String(int, unsigned char base = 10);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:65:18: note: String::String(unsigned char, unsigned char)
         explicit String(unsigned char, unsigned char base = 10);
                  ^
*** [.pio/build/esp8266-release/src/app.cpp.o] Error 1

fsck-block, 2023-03-08:
> Version / Git SHA:
> Version: 0.5.66

Tried to reproduce your issue but no problem to build the current stable (main) version 0.5.66 [esp8266-release]
(downloaded as zip-file)

deeplymbedded, 2023-03-09:
I added the -E option and took a look at the app.cpp.o and main.cpp.o file.
In both I can only find the usage of u_int16_t variable, but not the definition.
So it looks like somehow the header-file with the corresponding #define does not get included at all

Taking a deeper look at sys/types.h shows that the u_int16_t is encapsulated by a #ifdef __MS_types__ directive, which doesn't seem to be defined in my environment.
I try to dig deeper, currently it looks like really the target-system has something to do with it...

lumapu, 2023-03-09:
thanks for pointing that out

lumapu, 2023-03-09:
@deeplymbedded I checked the code again, both `main` and `development03` but didn't found any `u_int16_t` occurance. Does it come from our code our is it hidden in some library? Do you know a line number?

deeplymbedded, 2023-03-10:
Hello @lumapu , thanks for the feedback.
I traced it according to the error-description of the compiler:
In file included from hm/hmSystem.h:9:0,
from app.h:25,
from main.cpp:7:
hm/hmInverter.h:402:28: error: 'u_int16_t' has not been declared
String getAlarmStr(u_int16_t alarmCode) {
^

while compiling main.cpp it includes app.h, which includes hmSystem.h, which includes finally hmInverter.h 
There the u_int_16 is used in line 402.

I have also a Windows PC running, I will try to install platformio there and run the build to see what happens on that machine.

lumapu, 2023-03-10:
thanks, I will search for these places and try to resolve that. I don't know why here are these definitions were used.

deeplymbedded, 2023-03-10:
Thank you.
I think the main issue is that on my Linux system, platforio somehow uses different packages than on my Windows system.
I'm not that familiar with platformio, yet, hence I do not understand 100% how the IDE decides which libs to use.
The fact that I have different packages / libs installed on my Linux PC probably comes because I also had a tasmota project loaded there, which runs on an ESP8266, too and loaded some libs

lumapu, 2023-03-10:
just saw that the development version does not have this issue. Check Branch `development03`

deeplymbedded, 2023-03-10:
Yes you are right, the type-issue doesn't appear on the development03 branch.
However something is still not clean.
It seems like the tasmota project installed some packages via github URL which are modified and anyhow used by the ahoy project

Processing esp8266-release (platform: espressif8266; board: esp12e; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: g28bd99e
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2022.12.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.7.4+9 
 - tool-esptoolpy @ 1.40401.0 (4.4.1) 
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.1dde9cf
|   |-- ESPAsyncTCP @ 1.2.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ArduinoJson @ 6.20.1
|-- RF24 @ 1.4.5
|   |-- SPI @ 1.0
|-- Time @ 1.6.1
|-- espMqttClient @ 1.4.1+sha.15041c4
|   |-- ESPAsyncTCP @ 1.2.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- ArduinoJson @ 6.20.1
|-- Timezone @ 1.2.4+sha.7cf1425
|   |-- Time @ 1.6.1
|-- U8g2 @ 2.34.15
|   |-- SPI @ 1.0
|   |-- Wire @ 1.0
|-- GxEPD2 @ 1.5.0
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- Wire @ 1.0
|   |   |-- SPI @ 1.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 1.0
|   |   |-- SPI @ 1.0
|   |-- Wire @ 1.0
|   |-- SPI @ 1.0
|-- ESP8266WiFi @ 1.0
|-- Adafruit GFX Library @ 1.11.5
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 1.0
|   |   |-- SPI @ 1.0
|   |-- Wire @ 1.0
|   |-- SPI @ 1.0
|-- SPI @ 1.0
|-- DNSServer @ 1.1.1
|   |-- ESP8266WiFi @ 1.0
|-- LittleFS(esp8266) @ 0.1.0
|-- ESPAsyncTCP @ 1.2.2
|   |-- ESP8266WiFi @ 1.0
|-- Adafruit BusIO @ 1.14.1
|   |-- Wire @ 1.0
|   |-- SPI @ 1.0
|-- Hash @ 1.0
|-- Wire @ 1.0
Building in release mode
[... 600 lines cut ...]
/home/kja2li/.platformio/packages/framework-arduinoespressif8266@2.7.4+9/cores/esp8266/WString.h:70:18: note: String::String(float, unsigned char)
         explicit String(float, unsigned char decimalPlaces = 2);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266@2.7.4+9/cores/esp8266/WString.h:69:18: note: String::String(long unsigned int, unsigned char)
         explicit String(unsigned long, unsigned char base = 10);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266@2.7.4+9/cores/esp8266/WString.h:68:18: note: String::String(long int, unsigned char)
         explicit String(long, unsigned char base = 10);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266@2.7.4+9/cores/esp8266/WString.h:67:18: note: String::String(unsigned int, unsigned char)
         explicit String(unsigned int, unsigned char base = 10);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266@2.7.4+9/cores/esp8266/WString.h:66:18: note: String::String(int, unsigned char)
         explicit String(int, unsigned char base = 10);
                  ^
/home/kja2li/.platformio/packages/framework-arduinoespressif8266@2.7.4+9/cores/esp8266/WString.h:65:18: note: String::String(unsigned char, unsigned char)
         explicit String(unsigned char, unsigned char base = 10);
                  ^
*** [.pio/build/esp8266-release/src/main.cpp.o] Error 1
*** [.pio/build/esp8266-release/src/app.cpp.o] Error 1
=============================================================== [FAILED] Took 8.44 seconds ===============================================================

Environment      Status    Duration
---------------  --------  ------------
esp8266-release  FAILED    00:00:08.443

deeplymbedded, 2023-03-10:
I just removed all packages from esp8266 which came with the tasmota and loaded them newly, now build is working.
Anyway I wonder why they conflict, as I thought platformio is maintaining every package separately in the project context to avoid exactly such clashes

closed 2023-03-27 by lumapu