lumapu/ahoy #749: [ESP32] Enable Screensaver (pixel shifting) ohne Funktion in 0.5.95

soelli22, 2023-03-08:
### Platform

ESP32

### Model name

ESP32-D0WDQ5

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.95

### Github Hash

f5ef19c

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

bei der aktuellen Der Version funktioniert das Pixelshifting nicht. Als Display verwende ich ein SH1106.
Aktivieren und deaktivieren inkl. Reboot helfen nicht.
Weiterhin noch ein kleines kosmetisches Problem: Die Anzeige Reactive Power ist bei Total "var" und bei den einzelnen WR´s: "VAr"

DasUrmel, 2023-03-08:
Bestätige.
Das gilt auch für "Turn off while inverters are offline" - wird ebenfalls nicht beachtet.

lumapu, 2023-03-08:
bitte nochmal mit der neuen Dev `0.5.96` testen, falls nicht gefixt das `fixed dev` flag wieder entfernen =)

soelli22, 2023-03-09:
mit der 0.5.96 läuft Pixel Shifting wieder.

closed 2023-03-27 by lumapu