lumapu/ahoy #750: Welches esp32 Modul kaufen?!

KoLSMS, 2023-03-08:
Hallo,

sorry für die "blöde" Frage aber kann mir jemand einen genauen Link zu einem "ESP32 Dev Board NodeMCU WROOM32 WiFi" Modul schicken?

Bevorzugt Aliexpress...

Es gibt eine extrem große Auswahl und ich möchte vermeiden das falsche zu kaufen.

Vielen Dank

teletapi, 2023-03-08:
https://de.aliexpress.com/item/1005003818247483.html?spm=a2g0o.productlist.main.13.76816b0bKWV8a1&algo_pvid=eb9f7e3a-06b7-4b3f-a292-5b5e4d1ef442&algo_exp_id=eb9f7e3a-06b7-4b3f-a292-5b5e4d1ef442-6&pdp_ext_f=%7B%22sku_id%22%3A%2212000027263733299%22%7D&pdp_npi=3%40dis%21EUR%218.4%215.38%21%21%21%21%21%402100b88516782933676796310d06c2%2112000027263733299%21sea%21DE%21843218759&curPageLogUid=Nsb6xUwRE4Rq

teletapi, 2023-03-08:
Ich hab die von Berybase
https://www.berrybase.de/esp32-nodemcu-development-board

KoLSMS, 2023-03-08:
Hallo,

vielen vielen Dank, habe bei AliExpress zugeschlagen, kann geschlossen werden...

Sorry nochmals für die "dümmliche" Frage :-)

KoLSMS, 2023-03-08:
Erledigt

closed 2023-03-08 by KoLSMS