lumapu/ahoy #751: Ahoy zurücksetzen

Mart-Muc, 2023-03-08:
Hallo zusammen,

Ahoy kommt nichtmehr hoch nachdem ich die statische IP für NTP, DNS und Gateway eingegeben habe (nicht IP-Adresse)

Ich befürchte ich DEPP hab evtl das Subnetzmaske-Feld mit copy&paste erwischt. ???

Gibt's ne Möglichkeit das Teil zurückzusetzen ohne es neu zu flashen? (Denn es steht weit entfernt bei meiner Mutter)

(V5.66)

beegee3, 2023-03-08:
wenn du einen PC mit Wifi hast und in den AP Modus kommst (http://192.168.4.1), Passwort esp_8266, kannst du die Settings bearbeiten. Versuch es zunächst direkt. Klappt das nicht, Ahoy einmal rebooten (Strom unterbrechen), dann ist beim Neustart der AP Modus für mindestens 10 Sekunden verfügbar.
Viel Glück!

lumapu, 2023-03-08:
ansonsten kann man das auch per Fernwartung und der Projektseite https://ahoydtu.de/web_install und einem Chrome / Edge Browser lösen.

Mart-Muc, 2023-03-09:
Ja so werd ich es wohl machen müssen. Danke:-)

Nur so ins Blaue gedacht...
falls es mit 3 Zeilen Code gehen würde wäre das ein großes Enhancement:
Liese sich die "Factory Reset" Funktion der System-Seite nicht auch mit einem GPIO Kontakt (am besten >=10 sec) auslösen?

lumapu, 2023-03-09:
ich hatte eher an sowas gedacht:
abstecken, anstecken und > 5s warten, aber < 10s
das ganze 3x wiederholen, dann führt Ahoy einen Reset aus

knickohr, 2023-03-10:
Würde sich sowas auch mit einem der beiden vorhandenen Taster realisieren ?

Also z.B. beim Einstecken eine der Tasten für min. 10s drücken ?

Argafal, 2023-03-10:
Zur Inspiration: So macht es OpenMQTTGateway: Taster auf einem (beim Bauen der Software definierten) GPIO Pin. 5 Sekunden halten löst einen Reset der Einstellungen aus.

beegee3, 2023-03-10:
was ist denn der Vorteil zu einem abstecken, anstecken und innerhalb 10s im AP Modus anmelden?
Nach einem Factory Reset muss man doch sowieso in den AP Modus und alles neu einstellen.

lumapu, 2023-03-10:
> was ist denn der Vorteil zu einem abstecken, anstecken und innerhalb 10s im AP Modus anmelden?
> Nach einem Factory Reset muss man doch sowieso in den AP Modus und alles neu einstellen.

ist die Frage ob der gemeine Nutzer das in der kurzen Zeit schafft, meist sind es unbedarfte die von dieser Funktion Gebrauch machen sollen

Mart-Muc, 2023-03-10:
Ja, ich glaub meine 70 jährige Mutter (siehe Post 1) wird das in 10sec nicht schaffen… da wäre ein Zeitfenster von 60sec denk das Minimum. 

Deswegen würd ich auch nen resetknopf wie z.b. bei kommerziellen Zwischensteckern, etc vorziehn.

Könnte man dann <neue Idee> auch gleich als WPS- Schaltern nehmen.
Aber jetzt will ich mit neuen Ideen aufhören- denn ich denk es gibt bereits genug zu coden… ich schätze eure Arbeit- aber ich denk ihr solltet auch ein Privatleben haben ;-)

beegee3, 2023-03-10:
solange keine STA Verbindung (= Verbindung zum Router oder sonstigem AP) hergestellt wurde, müsste der AP Modus verfügbar sein. Erst mit einer STA Verbindung wird der AP Modus abgeschaltet (zumindest so die Theorie 😄). Bis dahin gibt es kein Zeitlimit. Die 10s sind die Wartezeit **bevor** eine STA Verbindung versucht wird. Wer also trotz funktionierendem STA eine AP Mode Verbindung will, der hat nur 10s dafür garantiert.

Mart-Muc, 2023-03-10:
Guter Tip, danke:-)
Na dann probier ich das am WE mal, und lass Mama erstmal die Fritz box ausschalten.

beegee3, 2023-03-10:
WLAN an der Fritzbox ausschalten reicht 😁

Mart-Muc, 2023-03-10:
Also..... die Fragestellung ändert sich:
Ich hatte keinen copy&paste Fehler, 
sondern der Ahoy hat sich von sich aus auf Werkseinstellung zurückgesetzt und alle Einstellungen waren weg.

wie kann das passieren? Hardwaredefekt? Trotz starken Netzteil (aber ohne Kondensator) in nem ganz ungünstigen Moment rebootet? Mehr fällt mir nicht ein.
Na mal schaun wie es morgen bei Sonnenschein weitergeht.

Danke nochmal

Mart-Muc, 2023-03-10:
Jetzt zweifel ich langsam an mir.
den einen hab ich wieder konfiguriert -passt soweit.

dann hab ich den anderen (v. 5.66, anderer wemos, gutes apple netzteil, mit 2x100uF, 200km andere Himmelsrichtung) über fritz box sitebysite vpn verbindung die Geokoordinaten angepasst (xx.xx) und schon verschwindet der. Nachdem er jetzt tagelang top funktioniert hat.

Auch werkseinstellung...

Wozu hab ich informatik studiert??? Ist heut mein Pechstag??
Ich geh jetzt ins Bett.

lumapu, 2023-03-10:
echt komisch, paar Tage noch, dann haben wir hoffentlich ein neues Release. Im Gegensatz zur 0.5.66 läuft die jetzige Dev Version meiner Meinung nach schon wesentlich stabiler.
Von selbst Settings verloren habe ich bisher noch nicht gehört. Bei Speichern von Einstellungen passiert es leider hin und wieder (je nach Menge der zu speichernden Daten), dass der JSON String zum Ende korrupt ist. Beim Laden wird das JSON trotzdem als valide empfunden, da wohl die Speicherroutine immer die schließenden Klammern einfügt und damit uU. die letzen Bytes einfach überschreibt wenn der Speicher ausgeht. Das führt dann beim Booten zu einer Exception. An dieser Stelle bin ich noch nicht ganz glücklich.

AlfonsKrismann, 2023-09-09:
Hallo, ich habe von der bei mir seit gut 7 Monaten stabil laufenden 0.566 Firmware (esp32 mit Display, Standardinstallation) auf 0.736 upgedatet. Update lief auch durch, Pin waren korrekt, alle Einstellung stimmten noch. die beiden HMs wurden noch als connected angezeigt, aber es kamen keine Daten rein... Dann habe ich das nrf24 aktiviert in den Settings und danach ging nichts mehr. Ich bekomme mit USB am Rechner nur serial port Fehler - auch nach Drücken von boot und kurz EN-Reset. Es wird auch kein Ahoy-AP für 10 sec aufgebaut. Dann kann man nichts mehr ausprobieren, oder? (Außer Neukauf).

lumapu, 2023-09-14:
hast du es über unseren Webinstaller auf https://ahoydtu.de schon probiert?

stefan123t, 2023-12-21:
@lumapu ist das ein Feature Request für eine Factory Default Settings Option oder ein (inzwischen gelöstes?) Update/Settings speichern Problem ?

lumapu, 2024-01-18:
dieser Feature Request zielt eher darauf ab, einen Ahoy zurückzusetzten ohne PC, also z.B. 5x kurz einstecken oä.

stefan123t, 2024-10-27:
Auch hier wäre eine Taste toll. Die Flash (natürlich nicht die Boot) Taste die auf den meisten Boards drauf ist kann man im Code nicht zufällig abfragen ?