lumapu/ahoy #755: Faulty irradiation calculation of CH4 - HM1500

nexulm, 2023-03-09:
Great update on the UI. 😀
—-
I believe there is an calculation error of the irradiation. For me at least for CH4. All modules are very similar orientated close together. Multiplicator of 10 for Ch4 would solve the issue for me!?!

[image]

beegee3, 2023-03-09:
please look into your setting for CH4. Is the Max Module Power value correct?

nexulm, 2023-03-09:
Good hint, there was a typo 4110 instead of 410 due to the fact that I've to re-add the inverter after updating from 0.5.92 to 0.5.93 this week. I will close the issue again although I can't reach the ESP8266 anymore after update, Save and PowerOn-Reset aswell. :-(

closed 2023-03-09 by nexulm