lumapu/ahoy #758: Ihr NRF24-Modul nicht erreichbar ist, überprüfen Sie die Verkabelung und Pinbelegung

Speedy231001, 2023-03-10:
Hi zusammen,

ich bin nach dieser Anleitung vorgegangen, allerdings bekomme ich immer folgende Fehlermeldung "Ihr NRF24-Modul nicht erreichbar ist, überprüfen Sie die Verkabelung und Pinbelegung" Dien Lötstellen sind alle sauber und die Verkabelung habe ich 2 x neu gemacht.

https://github.com/lumapu/ahoy/blob/main/Getting_Started.md#things-needed

Meine Hardware:

D1 ESP8266 Mini WLAN Board Mikrokontroller 

NRF24L01+ SMD Modul 2,4 GHz Wi-Fi Funkmodul 

Folgende Passage hat mich allerdings verwirrt:

"Das Funkmodul NRF24L01+ wird an die Standard-SPI-Pins angeschlossen:

SCLK (Signaluhr),
MISO (Master In Slave Out) und
MOSI (Master-Out-Slave-In)
Diese Pins müssen in der config.h konfiguriert werden."

Ich habe eine ready to flash datei genommen, sollte da nicht alles fertig konfiguriert sein?

Ich hoffe hier kann mich einer auf den richtigen Lösungsweg bringen......

Loetnase, 2023-03-10:
Welchen Softwarestand hast du denn überhaupt geflasht?  0.5.??
Wie ist diese Datei benannt? Vieleicht hast du die falsche

Speedy231001, 2023-03-10:
Hi,

Ich hab die 0.5.66 draufgemacht.... Das wemos an sich geht ja alles aber das nrf ist nich erreichbar... Ich hab die mit dem zusatz ssd geflasht und einmal die ohne den zusatz.

lumapu, 2023-03-10:
Dann gibt es immer den Tipp die Pins IRQ und CE zu tauschen, also Hardware und dann auch in den Settings.

mr-p666, 2023-03-10:
Hej,
bei meinem NRF-Modul mit Antenne hat vom Produktionsprozess etwas Zinn zwei der Connectoren gebrückt.
Wenn die Prüfung der Steckverbindungen also keinen Fehler ergibt, prüfe, ob irgendwelche Pins Durchgang haben.
[image]

lumapu, 2023-03-10:
allgemein ist eine 'optical inspection' des gesamten Moduls nicht falsch. Ich hatte auch schon ein Modul mit PCB-Antenne wo ein Bauteil eine Überlötung hatte und das Modul dadruch nicht funktonierte.

dtuuser, 2023-03-11:
Den Tipp von lumapu mal probiert?
Bei meinem D1 musste ich CE auf   D2 und IRQ  auf D1 umstecken und um konfigurieren. Hat irgendwas mit dem IQR an D4 und der Nähe zum G zu tuen

Speedy231001, 2023-03-11:
Hi,
Vielen Dank erstmal für die vielen Lösungsansätze..... Ich habe jetzt schon mal herausgefunden, dass ich ein Problem mit dem G habe..... Der lötpunkt macht Probleme.... Wenn icb das Kabel oben drauf halte läuft alles..... Sobald ich es auf den jumper stecke funktioniert es nicht mehr..... Ich überlege schon den jumper zu umgehen und das Kabel direkt drauf zu Löten....

lumapu, 2023-03-11:
das ist auch eines der bekannten Probleme: Steckkontakte.
Ich würde hier ganz klar empfehlen alles direkt zu verlöten, dann geht es auch über Jahre und korrodiert nicht

Speedy231001, 2023-03-12:
Hi,

nachdem ich jetzt "G" direkt mit dem Draht verlötet habe und die D3/D4 auf D1/D2 gesteckt und umkonfiguriert habe, läuft die DTU tadellos! 

Vielen Dank für die Tipps und die tolle Unterstützung!!!

closed 2023-03-12 by Speedy231001