lumapu/ahoy #759: [ESP8266] Display 0,96" SSD1306 geht nicht

ntfrnd, 2023-03-11:
### Platform

ESP8266

### Model name

https://www.makershop.de/plattformen/d1-mini/esp8266-mini-board/

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.97 dev

### Github Hash

28bd99e

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Ich habe ein 0,96" Display SSD1306 angeschlossen.
VCC an 3,3V
GND an GND
SCL an GPIO5 Pin D1
SDA an GPIO4 Pin D2

Im Setup habe ich den Displaytyp (SSD 1306 0,96") ausgewählt und Clock = D1 (GPIO5, SCL) und Data = D2 (GPIO4, SDA) gesetzt, gespeichert und rebootet.
ABER: Display bleibt dunkel. Nach Überprüfen der HW habe ich den Displaytyp geändert auf "SSH1106 1,3" ", Pinout unverändert, und siehe da: Das Display funktioniert.

[image]

[image]

knickohr, 2023-03-11:
Das ist so nicht ganz korrekt. Ja, der SH1106 kann als Workaround für das SSD1306 genommen werden, es kann aber sein das das Display dann komische Punkte am linken Rand anzeigt. Sollte so keine Dauerlösung sein !

gandalfred, 2023-03-12:
Display SSD1306 geht mit der 0.5.99 immer noch nicht. Mit dem Setting für SH1106 funktioniert das SSD1306 aber.

MiniOh, 2023-03-12:
Bei mir geht das SSD1306er mit SSD1306 Settings auch nicht, also es bleibt vollkommen schwarz.
Mit SH1106  im Setting gehts, sieht aber eben nicht sauber aus.

dixy245, 2023-03-12:
Ich hatte mal etwas gelesen, dass es programmiertechnisch einen Shift um 2 Pixel nach links beim SSD1306 bedarf. Ansonsten sollen die beiden Displays identish sein.

HorstyS, 2023-03-16:
Same here....ich hab eben die 0.5.100 aufgespielt und mein SSD1306 funktioniert nur mit der Auswahl SH1106. Pixelshift funktioniert auch ohne Probleme und alles scheint lesbar zu sein (bis auf die zu kleinen Statusinfos am unteren Rand).

HorstyS, 2023-03-16:
Merci, versuche ich gleich mal.
Frage: Was ist das rechte denn für ein Display und wie wird es angesteuert/konfiguriert? Sieht aus wie ein eInk :) Haben will ;)

HorstyS, 2023-03-16:
Jup, funktioniert nun auch mit der Einstellung SSD1316 und dem SSD1316 Display.
Well done

lumapu, 2023-03-16:
ja das rechte ist ein ePaper 1.54" 200x200px Waveshare per SPI angeschlossen 
geht nur mit ESP32

lumapu, 2023-03-16:
wichtig ist die V2 zu kaufen, es gibt verschiedene Merkmale dafür

knickohr, 2023-03-16:
V2 Rev 2.1 Am besten direkt bei waveshare.

Vorsicht, es gibt viele Anbieter die noch Restbestände „günstig“ vertickern. Es gibt mindestens 3 verschiedene Versionen mit teilweise unterschiedlichen Controllern. Es funktioniert nur das bisher neueste !

[image]

dtuuser, 2023-03-16:
> Dann nimm gleich mal die 101 !
> 
> damit funktioniert es wieder 😉
> 
> [image]

gibt es changelog zur 101?

knickohr, 2023-03-16:
Leider noch nicht gesehen. Da muß @lumapu noch Hausaufgaben machen 😇

dtuuser, 2023-03-17:
Okay. Danke.
Meine 5.92 läuft jetzt seit 12 Tagen 1 Std.

soebbing, 2023-03-18:
@knickohr Sorry für's de-railen, aber hast Du vielleicht einen Link zu dem Gehäuse? Das gefällt mir echt gut!

lumapu, 2023-03-20:
es gibt in Discord einen eigenen Channel dafür

closed 2023-03-27 by lumapu