lumapu/ahoy #760: Limit vom WR in zusätzlichem Test Ahoy

dtuuser, 2023-03-11:
Mal eine Frage zum Limit.  Wenn ich das Limit über CURL Aufruf setze, dann wird das beim genutzten Ahoy Stick in der Weboberfläche angezeigt. Da ich einen zweiten Ahoy Stick zum Testen neuer Versionen parallel nutze, ist mir aufgefallen, dass der 2te Stick immer 100% anzeigt. Ich dachte, das temporär gesetzte Limit wird auch im WR gespeichert

beegee3, 2023-03-13:
Das ist wie erwartet. Die WR senden nur auf Anfrage und nur an den Anfragenden. Wenn also ein ESP ein Limit setzt, bekommt er die Rückmeldung, dass die Anfrage bearbeitet wird. Um festzustellen, ob das Limit wie gewünscht gesetzt wurde, fragt der ESP direkt danach das im WR aktuell eingestellte Limit ab. Wiederum bekommt nur er den Wert mitgeteilt. Der zweite ESP bekommt von all dem nichts mit. Nach einem Reboot fragt ein ESP immer zuerst die aktuellen Daten des WR ab. So auch das aktuelle Limit.
Man könnte das Abfragen des aktuellen Limits natürlich periodisch machen, was aber normalerweise unnötig ist.
Deine Konstellation, dass mehrere Ahoy ESPs mit demselben WR kommunizieren, ist doch sehr speziell.
Zum Testen neuer Versionen geht das ok, solange der Test ESP nicht auch aktiv in die Limit Steuerung eingreift. Sonst wäre das sicher kontraproduktiv. Im Allgemeinen besteht keine Notwendigkeit, mehrere ESPs in dieser Form einzusetzen.

dtuuser, 2023-03-13:
Danke für die Detailerklärung. Dachte, das Limit wird immer mit abgefragt und ist Bestandteil des Streams.
Aktuell habe ich zwei Ahoys im Einsatz, um die Umstellung der Abfrage über api/inverter/id zu programmieren. Und dann versuche ich, die Auswertung der Daten vom WR, die Auswertung vom Tasmota Zähler und die Nulleimspeisungssteuerung in ein Programm zu packen. Aktuell machen das drei unabhängige Programme, da ich alle Werte auch noch mitlogge. Und im Sommer wird es dann interessant, wenn ich die Klimaanlage und Poolpumpe über Hue ansteuer, anstatt den WR zu drosseln. Aber für  diese Tests brauche ich dann viel Sonne am Stück

closed 2023-03-25 by lumapu