lumapu/ahoy #761: 0.5.98 keine Regelung mehr über MQTT

jimknopf63, 2023-03-11:
### Platform

ESP32

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.98

### Github Hash

3e919b4

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Ich beobachte das bei der 0.5.98 die Regelung per MQTT nicht mehr funktioniert. In Ahoy werden nach Limitierung enorme Prozentwerte weit über 100% angezeigt und der WR bleibt schön bei 100%
[image]
[image]

lumapu, 2023-03-11:
über welches subscribe limitierst du den Wechselrichter?
Seit `0.5.98` muss man zB. `/inverter/Ctrl/limit/0` mit einem Wert zwischen 2-100 für ein relatives Limit oder einen Wert von 0W-65535W für ein absolutes Limit senden. Ich habe das im chancelog erwähnt und das User Manual angepasst

jimknopf63, 2023-03-11:
> über welches subscribe limitierst du den Wechselrichter?
> Seit `0.5.98` muss man zB. `/inverter/Ctrl/limit/0` mit einem Wert zwischen 2-100 für ein relatives Limit oder einen Wert von 0W-65535W für ein absolutes Limit senden. Ich habe das im chancelog erwähnt und das User Manual angepasst

Ich hatte bisher mit Hoymiles/ctrl/limit_nonpersistent_absolute/0' geregelt und nach dem Update mit Hoymiles/ctrl/limit/0 .
Trotzdem stimmt es irgendwie nicht. Er übernimmt quasi meine Watt Limitierung auf % wie mir scheint

dtuuser, 2023-03-11:
Das bedeutet, dass ich das Limit über CURL auch anders setzen muss?
Bis jetzt so...

curl -i -X POST -H "Content-Type: application/json" -d "{"id":"0","cmd":"limit_nonpersistent_absolute","val":"191"}" http://192.168.0.232/api/ctrl

Und jetzt so?

curl -i -X POST -H "Content-Type: application/json" -d "{"cmd":"http://192.168.0.232/inverter/ctrl/limit_nonpersistent_absolute/0  191W"}"

beegee3, 2023-03-11:
der richtige Link zum Changelog ist https://github.com/lumapu/ahoy/blob/development03/User_Manual.md
Die Angabe "%" oder "W" hinter dem Wert ist für **MQTT**.
Allerdings ist da bei der Beschreibung für **Power Limit absolute (non persistent) [Watts]** das Beispiel noch falsch, statt `inverter/ctrl/limit_nonpersistent_absolute/0` muss es `inverter/ctrl/0` heißen (@lumapu bitte korrigieren).

hier die richtigen Beispiele:

**Power Limit relative (non persistent) [%]**
inverter/ctrl/limit/0     70
_ist dasselbe wie_
inverter/ctrl/limit/0     70%

**Power Limit absolute (non persistent) [Watts]**
inverter/ctrl/limit/0     600W

Für **curl** gilt nach wie vor die Beschreibung von **Control via REST API**, hat sich m.E. nicht geändert.

jimknopf63, 2023-03-11:
> der richtige Link zum Changelog ist https://github.com/lumapu/ahoy/blob/development03/User_Manual.md Die Angabe "%" oder "W" hinter dem Wert ist für **MQTT**. Allerdings ist da bei der Beschreibung [...]

Danke habe es geändert und sende nun W direkt im Anschluss ohne Leerzeichen, jetzt funktioniert es. Kann geschlossen werden. Mein Fehler

closed 2023-03-11 by jimknopf63