lumapu/ahoy #762: ESP32 Dev Board NodeMCU WROOM32 WiFi with differently labelled pins

thorag76, 2023-03-11:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

Can't recognize it on the chip, it's way too small even in a picture of cell phone, but i ordered "NRF24L01+ PLUS - PA LNA SMA Antenne Long Range Funk Modul Raspberry Arduino" in makershop.de together with the " Espressif ESP32 WLAN Dev Kit Board Development Bluetooth Wifi v1 WROOM32 NodeMCU"

### Antenna:
* [ ] circuit board
* [x] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

Not using a capacitor and I am not sure if the WROOM32 board has one or does need one.

### Connection diagram:
* [ ] Image of the your wiring attached
* [x] this is exactly my problem, none of your images fit

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU
* [x] not yet started, but I'll start with ahoy

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [x] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
* [x] not yet started

I haven't got that far - not even beyond wiring, the problem is:
None of your diagrams fit. 
My WROOM32 has different labels. They look exactly like those here: 
https://europe1.discourse-cdn.com/arduino/optimized/4X/8/2/6/8264242978c38d1dcbece30e048f1e16f9bd1554_2_612x500.jpeg 
The image is taken from https://forum.arduino.cc/t/connecting-2-esp32-wroom-with-nrf24l01-radio-module/1056258

If I compare them to your connection diagram your wroom32 has "G<number>" labels whereas mine has "D<number>" labels and they arrangement is totally different.

I have no idea how to wire the NRF24L01+ with the WROOM32 board and I have no idea what I will have to enter in config files for this

lumapu, 2023-03-11:
here are good Diagrams of different ESP32:
https://www.studiopieters.nl/esp32-pinout/

You got a 30 pin version of ESP32. You only have to compare the GPIO numbers, eg. GPIO25 is printed on your boards as D25.

thorag76, 2023-03-12:
Hey, thanks a lot for your very quick reply and for your help :)

I followed your advice, connected all numbers according to the diagram here in the ahoy project (last diagram). 

The only connection I am unable to do is the green one, which connects G16 to the IRQ of the NRF24...

I have no D16. From the page you send me, this should the pin labeled "RX2"

Can I leave it out safely or should I just connect it to RX2 and go with the recommended config settings?

lumapu, 2023-03-12:
sure, connect to RX pin and go with the recommend settings

thorag76, 2023-04-02:
Hey, 

thanks a lot for your help - it worked right the first boot!

Do you still recommend an additional capacitor even if the module with external antenna and the "bigger" ESP32 is used?