lumapu/ahoy #765: Bug Offical 5.66 - WEB Oberfläche nicht mehr erreichbar

theunknownstarman, 2023-03-12:
## Hardware
  * ESP8266

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
circuit board

## Software
AhoyDTU

### Version / Git SHA: 
Git SHA: f8fe044 :: 0.5.66

### Desktop OS:
Mac OS / Safari Browser

Das Board wird über ein stabiles WLAN versorgt.

Fehler:
**Die Web Oberfläche stürzt sporadisch mehrmals am Tag ab.**
Nach WLAN reconnect funktioniert die WEB Oberfläche dann wieder.
Die Daten vom Hoymiles werden trotz WEB Oberflächen Absturtz abgeholt und via MQTT an Homeassistant gesendet. 

Ein Reflash und ein Factory Reset (5.66er) brachte keine Lösung.
Testweise wurde die 5.99 gelashed. Damit gab es massive andere Probleme.

Hat jemand einen Tip?

tqma1, 2023-03-12:
Die von mir bisher favorisierten Development Versionen sind die 0.5.77 und die wesentlich frischere 0.5.93
Mit 0.5.99 verliert meine Ahoy DTU (ESP8266) immer wieder sämtliche Einstellungen.

theunknownstarman, 2023-03-12:
Dank Dir ...

... woher bekomme ich die 0.5.77 od. 0.5.93?

tqma1, 2023-03-12:
Dort wo Du auch die 0.5.99 heruntergeladen hast …
[https://github.com/lumapu/ahoy/actions/runs/4322762858](url)

niekniek89, 2023-03-13:
I have this issue with 0.5.66 as well as all other DEV releases to date (0.5.97).
As soon as I go to the web page several times a day, eventually (after +-1 day) the ahoydtu crashes, and restarts (and sometimes faster than 1 day).

As long as I do NOT go to the web page, and only look in home assistant (MQTT), it works fine for days.

Or the ESP8622 is not sufficient, or are there still bugs in the software?

niekniek89, 2023-03-13:
unfortunately also with 0.5.99 the web page crashed.
I get new data every 10 seconds. would that be too fast (default is 30).

tqma1, 2023-03-13:
I do not have problems with my ESP8266 and i am getting new Data every 5 Seconds.
i am now on 0.5.96 - 0.5.99 was not running stable on my system.

niekniek89, 2023-03-13:
@tqma1 

can you share 0.5.96 with me by chance? I can't download it anywhere anymore.

didn't this release crash for you then?
I've had this with all dev versions so far, but haven't tried .96 yet.

tqma1, 2023-03-13:
you can find it here ...

[https://github.com/lumapu/ahoy/actions/runs/4369201523](url)

niekniek89, 2023-03-13:
@tqma1 
I can't find any downloads at your url.
[image]

tqma1, 2023-03-13:
I have seen now ...
you find it under "Actions":
[added Nokia display again for ESP8266 #764](https://github.com/lumapu/ahoy/actions/runs/4369201523)

niekniek89, 2023-03-13:
thanks, i have installed 0.5.96 now.
hopefully that works better.

Argafal, 2023-03-13:
I believe this is a duplicate of #660. Maybe we closed it a little bit too soon :) Can those of you who reported problems in this issue here please have a look at the screenshot in #660: is it the same problem you describe here or a new different one?

@niekniek89 I think I said it to you in the Hoymiles tweakers forum a few days ago: the webserver failing is a known bug present in all recent versions including 0.5.66. It's not a bug that's caused by a "wrong" piece of code somewhere, in a sense that would be easier to fix. It is caused by a non-optimal use of the heap memory. If you look in /system, the heap fragmentation value is too high. It's a percentage and it should be close to zero, maybe a few percent, but not around 40-50%). Some Ahoy versions show the issue more than others, but all are currently affected by the underlying root cause. That is also why you see it running more or less stable when you don't use the webUI. It's the same symptoms here for me. I agree it's very unfortunate. A lot of efforts are made from different persons to mitigate the heap fragmentation :) It will just take some time to resolve. 

Running on an ESP32 should not have this problem. It should affect only ESP8266s.

niekniek89, 2023-03-14:
since I installed 0.5.96, the "heap" usage seems to be reduced.
it has now been running for 19 hours... no idea if this will still shoot up as soon as there is more sun?

I will also keep an eye on github, and thank you very much for all the work / support! I have patience!

[image]

Argafal, 2023-03-25:
Könnten Betroffene (und auch Nicht-Betroffene) bitte kurz diese Fragen beantworten:

- Version?
- Wie viele Inverter und welche?
- Inverter-Abfrage Interval aus den Einstellungen? (Default: 30)
- MQTT Server konfiguriert: ja/nein?
- MQTT-Interval aus den Einstellungen? (Default: 0)
- WebUI aktiv benutzt: ja/nein?
- Typische heap_fragmentation Wert nach 20 Minuten Laufzeit? Siehe /system
- Regelmässige Probleme dass die WebUI kaputt geht, ja/nein? (siehe screenshot in issue https://github.com/lumapu/ahoy/issues/660)

Argafal, 2023-03-27:
Neues release 0.6.0 mit vielen Verbesserungen. Kann jemand das WebUI Problem mit ahoy 0.6.0 reproduzieren?

dm-zwoelf, 2023-03-28:
- ESP8266 oder ESP32? ESP8266
- Version? 0.6.0
- Wie viele Inverter und welche? 3 * HM-1500
- Inverter-Abfrage Interval aus den Einstellungen?  30
- MQTT Server konfiguriert: ja/nein? ja
- Falls ja, auf welcher Hardware läuft der MQTT Server?  Raspberry PI 4
- MQTT-Interval aus den Einstellungen? (Default: 0) 0 
- WebUI aktiv benutzt: ja/nein? ja
- Typische heap_fragmentation Wert nach 20 Minuten Laufzeit? 35
- Regelmässige Probleme dass die WebUI kaputt geht, ja/nein? ja

theunknownstarman, 2023-03-30:
ESP8266
Version 0.6
1 Hoymiles 1500
(Default: 30)
MQTT: ja
MQTT als AddOn unter Homeassistant (Synology VM)
MQTT-Interval (Default: 0)
WebUI aktiv benutzt: ja
Typische heap_fragmentation: 4
Regelmässige Probleme dass die WebUI kaputt geht:

Keine Probleme mit mit WLAN disconnects bzw. WebUI Erreichbarkeit.

Es gab ein Problem vom Update von Nightly Built 106 auf 0.6

Die WebUI zeigte keinerlei Settings/Live usw. mehr. 
Das Problem ließ sich nur mit einem "harten" Reboot (mehrmals Stromstecker ziehen) beheben.

plasticmirror, 2023-06-03:
Hi bei mir sieht es ähnlich zu Issue #660 aus. Der Webserver stürzt sogar teilweise ab, wenn man nur die Seite neu laden möchte. Auch im nicht eingeloggten Zustand tritt das Problem auf.

ESP8266
Version 0.6.9
1 x HM1500
Inverter-Abfrage Interval aus den Einstellungen? (Default: 30)
MQTT Server konfiguriert: nein
MQTT-Interval aus den Einstellungen? (Default: 0)
WebUI aktiv benutzt: ja
Typische heap_fragmentation Wert nach 20 Minuten Laufzeit: 4
Regelmäßige Probleme dass die WebUI kaputt geht: ja

Chris0706, 2023-07-07:
Have the same issues. Webserver completely broken, but MQTT reporting fine.

ESP8266
Version 0.6.9/0.7.x (current dev build) 
1 x HM800
Inverter-Abfrage Interval aus den Einstellungen? (Default: 30)
MQTT Server konfiguriert: ja
MQTT-Interval aus den Einstellungen? (Default: 0) 0
WebUI aktiv benutzt: ja
Typische heap_fragmentation Wert nach 20 Minuten Laufzeit: kann nicht mehr abgerufen werden. 
Regelmäßige Probleme dass die WebUI kaputt geht: ja

lumapu, 2023-08-05:
bitte nochmal mit dem neusten Release `0.7.25` testen

plasticmirror, 2023-08-12:
Mit dem Release 0.7.26 nach erstem Test keine erkennbaren Probleme mehr. 
Ich beobachte und hoffe bzw. denke, dass es so bleibt.

Woran könnte es gelegen haben?

lumapu, 2023-08-13:
es liegt wie immer leider am heap des ESP. Dieser ist sehr knapp und ging in früheren Versionen immer mal wieder aus.
An dieser Stelle wieder viel optimiert, sodass das gesamte System jetzt stabiler läuft.

theunknownstarman, 2023-08-13:
Im Moment kann ich mir nur damit behelfen, dass ich die nächtliche Ahoy Reboot Funktion verwende um den heap nicht an die Grenze ansteigen zu lassen. 
Mit der Official Vorgänger Version war das nicht notwendig.

lumapu, 2023-08-13:
ok, interessant. Ein Problem ist noch bzgl MqTT bekannt, das will ich demnächst bearbeiten.
Ich kann mich über die Stabilität der aktuellen Dev Version nicht beklagen, allerdings geht die heap frag nicht mehr unter 25. Das liegt wohl an dem `last_succcess`

closed 2023-09-16 by lumapu

plasticmirror, 2023-09-17:
Update zur Info:
das System läuft nun mit der 0.7.26 seit mehreren Wochen ohne reboot durch und die Oberfläche macht was sie soll. MQTT habe ich allerdings nicht konfiguriert.