lumapu/ahoy #767: API über Passwort schützen -Admin Password Protection for API

Mart-Muc, 2023-03-13:
V 5.66

Hallo zusammen,
Ich habe gerade gemerkt, dass ich über die api ohne Passwort zugreifen kann. (beide 'außerhaus' DTUs funktionieren übrigens wieder:-)
Ich habe ein Admin Password gesetzt und standardmäßig nur die Live-View ohne gelassen, sodass die Fritz box ne Portweiterleitung machen kann.

Habe nun "Angst", das ich da nun ein offenes Tor habe und mir (im unwahrscheinliche Fall eines Druckverlustes;-) jemand das Limit, etc. verstellen kann.

Ist es möglich im Protection Tab auch die Api hinzuzufügen?

Oder ist das in der Dev-Version schon längst gemacht? Dann sorry...und bitte diesen Report schließen.

Danke, Mart

lumapu, 2023-03-13:
Port Weiterleitungen sind verboten! Ich sage es so krass auch wenn es nicht wirklich verboten ist, aber das selbst zum kurzen debuggen schon zu gefährlich.
Bitte nur über VPN zugreifen

Die API wird an sehr vielen Stellen von der Web UI gebraucht. Daher steht sie noch offen wie ein Scheunentor. Früher oder später muss man zumindest die Wechselrichter betreffenden Teile sperrbar machen

Mart-Muc, 2023-03-13:
Danke, das dacht ich mir schon. 
Wäre nur für den einen Standort ohne Smarthome/Raspi/HomeAssistant gewesen

Warnung in der Readme.md oder besser Settings/Protection Tab wäre toll, 
z.B mit ausgegrauten Kasterl in der Liste, sodass man sieht, dass man es nicht aktivieren kann -> und kann ja dann in einer zukünftigen Version dann änderbar werden.

Schönen Abend, Mart

lumapu, 2023-03-13:
guter Input 😊👍

lumapu, 2024-08-16:
API ist seit einigen Versionen Passwortgeschützt sobald ein admin Passwort gesetzt wurde.

closed 2024-08-16 by lumapu