lumapu/ahoy #771: [ESP8266] Einstellungen gehen verloren

eupesco, 2023-03-14:
### Platform

ESP8266

### Model name

ESP8266MOD 12F

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.100

### Github Hash

6da58e3

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

1.3" OLED SH1106 

### Error description

Nach einen Powercycle vom ESP8622, gehen die Einstellungen verloren (d.h. auch Wifi, Inverter usw; komplett). 

"your settings are invalid."

Argafal, 2023-03-14:
I can reproduce the reported issue with 0.5.99.

tqma1, 2023-03-14:
Kann ich so nicht bestätigen - auch nicht wenn ich die Spannungsversorgung ab- und wieder an-stecke.

T-Paul80, 2023-03-14:
Ich kann es leider bestätigen, gleiches Verhalten in meinem Fall mit der 0.5.101. 
Über AP vorgenommene WiFi Einstellungen werden zunächst übernommen und bleiben nach dem Neustart erhalten. Nachdem ich dann meine Wechselrichter, mqtt, Display und Standort etc. eintrage, gehen alle Einstellungen nach speichern -> Neustart verloren.

Argafal, 2023-03-15:
> Ich kann es leider bestätigen, gleiches Verhalten in meinem Fall mit der 0.5.101. Über AP vorgenommene WiFi Einstellungen werden zunächst übernommen und bleiben nach dem Neustart erhalten. Nachdem ich dann meine Wechselrichter, mqtt, Display und Standort etc. eintrage, gehen alle Einstellungen nach speichern -> Neustart verloren. Nachtrag: Problem tritt nur auf, wenn ich einen zweiten Wechselrichter eintrage.

@T-Paul80: Da beschreibst du jetzt aber etwas anderes als der Originalposter dieses Issues ;)

For the record: Ich kann @T-Paul80's Problem genau reproduzieren. Beim zweiten Setzen (Nachkorrigieren) von Einstellungen gehen auch bei mir die Einstellungen komplett verloren, und ich fange wieder via Ahoy's eigenem Access Point an. @eupesco's Beschreibung, dass das bei einem Powercycle passiert, klingt ähnlich von den Symptomen her. Allerdings kann ich das (Powercycle->Einstellungsverlust) selbst nicht reproduzieren.

eupesco, 2023-03-15:
Ich checks am Freitag nochmal. Ich habe zwei Wechselrichter, könnte sein, dass es damit zusammenhängt.

Argafal, 2023-03-15:
For the record: Das Problem das ich mit 0.5.99/0.5.100/0.5.101 sehe, tritt auch mit 0.5.96 auf. Gehe ich auf 0.5.93 zurück (git hash https://github.com/lumapu/ahoy/commits/71ea715) kann ich die Einstellungen wieder erfolgreich speichern. Ein erster Anhaltspunkt, das kann natürlich auch Zufall sein. 

Ich konfiguriere wie folgt:
- System: Die GPIO pins für NRF und den NRF Power Level. Keine LED, kein Display, die bleiben Default.
- Netzwerk: SSID und Passwort
- Protection: alles auf Default
- Inverter: zwei Inverter, ein HM-600, ein HM-1500. Jeweils mit "max power module" Werten. Interval: 10, und all drei Reset ein.
- NTP Server: default
- Sunrise/Sunset: Latitude, Longitude, Offset 60 minutes, Stop polling enabled.
- MQTT: Server IP. Rest Default.
- Display Config: unverändert.

Reboot upon save = enabled, press Save button.

Vielleicht hilft meine Liste ja, das zu reproduzieren.

lumapu, 2023-03-15:
ich werde es testen

Argafal, 2023-03-16:
As promised I went to try to find the offending commit. a4799b497ddc6935a9cfbfc80bbead6d475a95a8 works. The one right after (f7a7026ac88f161ad02c84e5eb90d56695124805) does not. I tried to review that commit. The settings routines are majorly reworked, so it's a good candidate for the origin of the problem. 

However, I must be doing something wrong. For your amusement I will share what it looks like for me, and hopefully someone can help me understand better.

```
 git diff f7a7026ac88f161ad02c84e5eb90d56695124805 a4799b497ddc6935a9cfbfc80bbead6d475a95a8|wc -l
10385
```
To me this looks like a gigantic commit touching 10385 lines. Here is list of things the `git diff` touches upon: the build environment, the readme files, the setup routines of the main app, the main communication loops, the mqtt communication, interrupt handling, displays, settings configuration, default pin configurations, dark mode, the size of the settings json, inverter data and communication, a circular buffer for the radio communication, MI payloads, more display routines, logging routines, html and css. 

The commit message is "renamed var to VAr". So I hope this gave you a good chuckle and someone can help me understand what's going on :)

Argafal, 2023-03-16:
For the record: 71ea7155ca2b6d16055a55d1c303a048698fe8bc also works. That's Fri Mar 3 12:01:19 2023 +0100, one of the 0.5.93 hot fixes.

lumapu, 2023-03-16:
thank you for pointing it out. I think you can make diffs between two commits if you pass two SHAs to `diff`.
Maybe there is an option to hide whitespace chances to further reduce number of changes during compare.
I really like the windows tool `TortoiseGit`

Argafal, 2023-03-17:
I am walking through the commits. f5ef19ccab0f9c8bc68d339a72d4e19f0c248c05 doesn't work for me, settings get lost. https://github.com/lumapu/ahoy/commit/71ea7155ca2b6d16055a55d1c303a048698fe8bc works for me, stable and fine. In between are 36 commits.

knickohr, 2023-03-17:
Das wäre dann die 93, danach kam das ePaper 😲

Argafal, 2023-03-18:
Ja, das scheint das Problem für mich zu lösen. Toll gemacht, @lumapu. Ach wie wunderbar :)

@eupesco Kannst du 0.5.102 ausprobieren bitte? Löst das für dich auch das Problem mit den Einstellungen?

T-Paul80, 2023-03-18:
Anmerkung, ich habe zunächst auf die .102 upgedatet, dann Display angepasst, soweit alles funktioniert. 
Nach dem Neustart gemerkt, dass Device Host Name leer ist. Also erneut "AHOY-DTU" eingetragen und Checkbox bei "Dark Mode" gesetzt. Nach dem Neustart waren die Einstellungen, bis auf WiFi , wieder weg.

eupesco, 2023-03-18:
@Argafal: würde sagen, läuft wieder. Danke, @lumapu

tqma1, 2023-03-18:
> Das wäre dann die 93, danach kam das ePaper 😲

Nur für den ESP32 oder auch für den ESP8266?

knickohr, 2023-03-18:
ePaper gibt’s nur für den ESP32

tqma1, 2023-03-18:
Schade, da gibt es so ein nettes Set (ESP8266 und 1,54“ ePaper) ….

closed 2023-03-27 by lumapu