lumapu/ahoy #773: 0.5.100 Yield Day auf 0 nach Inverter "Neustart"?

MiniOh, 2023-03-14:
## Hardware
  * [X] ESP8266
  * [X] ESP32
  * [ ] Raspberry Pi

Hallo, 

ich habe ein paar Mal das Verhalten beobachtet, dass Yield Day auf 0 gesetzt wird, wenn der Inverter innerhalb eines Tages offline geht und dann wieder den Betrieb aufnimmt.
Ist das ein Verhalten, welches vom Inverter selbst kommt, oder etwas was man mittels DTU beeinflussen kann.

Ausgelöst wurde das in den vergangenen Tagen ein paar Mal extrem starke Bewölkung.
Wenn **ein** Inverter alleine in Betrieb ist wird dadurch auch die **Total Yield Day** auf 0 gesetzt, 
bei meheren Inverten reduziert sich eben nur die Total Yield Day durch die Inverter Yield Day´s, welche offline waren.

Könnt Ihr das bestätigen, bzw. gibt es hier Ideen für einen Workaround?

Danke euch schon mal ;)

mroenne2022, 2023-03-14:
Hallo @MiniOh : 
Ich hab zwar im Moment gerade OpenDTU mit einem ESP32 im Einsatz , aber genau das Verhalten hatte ich in den letzten Tagen auch 2 mal. Es passiert genau so wie du beschrieben hast , wenn kurz vor "Feierabend" der WR wegen starker Bewölkung offline geht und wenn es dann kurz wieder aufhellt und der WR nochmal wieder kommt - dann kam dieser Fall. Morgens am nächsten Tag ist dann wieder alles OK.  Mich stört es nicht so sehr weil ich keine weiteren Verarbeitungen der Daten hab. Ich nutze lediglich die Weboberfläche.

lumapu, 2023-03-14:
wie ist bei euch die Einstellung?

[image]

knickohr, 2023-03-14:
Das hatte ich die letzten Abende mehrmals. Bei mir läuft parallel noch ein Tagesbetriebsstundenzähler. Und vor der Abends nur ein paar Minuten anzeigt, dann hatte der Tag wieder mal 2 oder mehr Solartage 🤪

beegee3, 2023-03-14:
Der Inverter beginnt nach einer Offlinephase immer mit einem YieldDay Wert 0. Auf wenn Ahoy noch den "wahren" YieldDay Wert hat, nützt es nichts, denn der Inverter meldet den 0 Wert. Und der wird von Ahoy so übernommen.
Man könnte versuchen, programmtechnisch entgegen zu steuern, aber @lumapu tu dir das nicht an, im Detail wird das wieder total kompliziert, auch wegen der schon vorhandenen 'Reset values'. Ist vielleicht was für Version 0.7.xx 😬

MiniOh, 2023-03-15:
> 

OK, habe ich mir fast so gedacht, dass hier einfach der Inverter ne 0 meldet.
Aber ja, ich gebe euch recht, das ist sicherlich den Aufwand nicht wert.
Man kann ja auch einfach mit dem Total Yield arbeiten und den Tageswert berechnen, dann passt das auch.

closed 2023-03-25 by lumapu