lumapu/ahoy #775: version 0.5.100 Import JSON Settings failure

beegee3, 2023-03-14:
der Import funktioniert nicht, im Serial Debug: `W:failed to load json, using default config`

In `web.h`, `onUpload2` im Bereich `if (final)` ein paar Änderungen (u.a. fehlte ein '/' bei `readSettings`), dann geht's. Habe noch ein `LittleFS.remove("/tmp.json");` eingefügt, ist wohl nicht notwendig, find ich aber sauberer.
Wenn's damit ok ist, dieses Issue bitte direkt wieder löschen 😄
```
        void onUpload2(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
            if (!index) {
                mUploadFail = false;
                mUploadFp = LittleFS.open("/tmp.json", "w");
                if (!mUploadFp) {
                    DPRINTLN(DBG_ERROR, F("can't open file!"));
                    mUploadFail = true;
                    mUploadFp.close();
                    return;
                }
            }
            mUploadFp.write(data, len);
            if (final) {
                mUploadFp.close();
                char pwd[PWD_LEN];
                strncpy(pwd, mConfig->sys.stationPwd, PWD_LEN); // backup WiFi PWD
                if (!mApp->readSettings("/tmp.json")) {
                    mUploadFail = true;
                    DPRINTLN(DBG_ERROR, F("upload JSON error!"));
                } else {
                    LittleFS.remove("/tmp.json");
                    strncpy(mConfig->sys.stationPwd, pwd, PWD_LEN); // restore WiFi PWD
                    mApp->saveSettings(true);
                }
                if (!mUploadFail)
                    DPRINTLN(DBG_INFO, F("upload finished!"));
            }
        }
```

knickohr, 2023-03-17:
101 hat das gleiche Problem, egal ob 8266 oder der 32er 😪

beegee3, 2023-03-17:
wurde auch noch nicht korrigiert

closed 2023-03-27 by lumapu