lumapu/ahoy #779: VSCode + Platform IO: Quellcode seit einigen Versionen nicht mehr kompilierbar

ralfsfb, 2023-03-16:
Hallo zusammen,

seit einigen Versionen kann ich den Quellcode leider nicht mehr fehlerfrei übersetzen. Es erscheinen immer Fehlermeldungen, wie z.B.

fatal: not a git repository (or any of the parent directories): .git
CalledProcessError: Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.:
  File "C:\Users\ralf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(item, exports="env")
usw.

wenn ich älteren Code (so bis Version 0.5.89) übersetzen möchte, funktioniert dies problemlos.
Hat jemand Tipps, welche Einstellungen geändert werden müssen, um wieder übersetzen und auf den ESP32 übertragen zu können?

Vielen Dank für eure Tipps im Voraus!
LG ralf

ralfsfb, 2023-03-16:
Hallo lumapu,

vielen Dank für deine schnelle Antwort.
Ja, ich habe bisher immer das zip-Archiv heruntergeladen, entpackt und dann in VSCode geöffnet, um für das 0,97" OLED noch ein paar kleine Anpassungen vorzunehmen.

Habe nun das Archiv nochmal in einen leeren Ordner entpackt, im Terminalfenster deinen Befehl eingegeben und ausgeführt.

PS C:\Users\ralf\Documents\PlatformIO\Projects\ahoy-development03\src> git clone https://github.com/lumapu/ahoy
Cloning into 'ahoy'...
remote: Enumerating objects: 8660, done.
remote: Counting objects: 100% (686/686), done.
remote: Compressing objects: 100% (297/297), done.
remote: Total 8660 (delta 402), reused 621 (delta 363), pack-reused 7974
Receiving objects: 100% (8660/8660), 20.54 MiB | 23.82 MiB/s, done.
Resolving deltas: 100% (5716/5716), done.
PS C:\Users\ralf\Documents\PlatformIO\Projects\ahoy-development03\src> 

Beim Kompilierungsversuch erschienen dann leider wieder folgende Meldungen:

 *  Task wird ausgeführt: C:\Users\ralf\.platformio\penv\Scripts\platformio.exe run --target clean --environment esp32-wroom32-release 

Processing esp32-wroom32-release (platform: espressif32; board: lolin_d32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: g0000000
fatal: not a git repository (or any of the parent directories): .git
CalledProcessError: Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.:
  File "C:\Users\ralf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(item, exports="env")
  File "C:\Users\ralf\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\ralf\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\ralf\Documents\PlatformIO\Projects\ahoy-development03\src\web\html\convert.py", line 145:
    convert2Header(val, version)
  File "C:\Users\ralf\Documents\PlatformIO\Projects\ahoy-development03\src\web\html\convert.py", line 85:
    data = htmlParts(inFile, "includes/header.html", "includes/nav.html", "includes/footer.html", version)
  File "C:\Users\ralf\Documents\PlatformIO\Projects\ahoy-development03\src\web\html\convert.py", line 57:
    link = '<a target="_blank" href="https://github.com/lumapu/ahoy/commits/' + get_git_sha() + '">GIT SHA: ' + get_git_sha() + ' :: ' + version + '</a>'
  File "C:\Users\ralf\Documents\PlatformIO\Projects\ahoy-development03\src\web\html\convert.py", line 12:
    return subprocess.check_output(['git', 'rev-parse', '--short', 'HEAD']).decode('ascii').strip()
  File "C:\Users\ralf\.platformio\python3\lib\subprocess.py", line 424:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\ralf\.platformio\python3\lib\subprocess.py", line 528:
    raise CalledProcessError(retcode, process.args,
=================================================================================================== [FAILED] Took 1.30 seconds ===================================================================================================

Environment            Status    Duration
---------------------  --------  ------------
esp32-wroom32-release  FAILED    00:00:01.301
============================================================================================= 1 failed, 0 succeeded in 00:00:01.301 ============================================================================================= 

 *  Der Terminalprozess "C:\Users\ralf\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'clean', '--environment', 'esp32-wroom32-release'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. 

Irgend etwas scheine ich also noch falsch zu machen ...

beegee3, 2023-03-16:
habe selbst eine Entwicklungsumgebung, die ohne Internetverbindung auskommt. Dort benutze ich den zip Code.
Ersetze mal in convert.py die [Zeile 12](https://github.com/lumapu/ahoy/blob/3e67f2a52f26d5ef6f1eafe0ca5f0544a6b92281/src/web/html/convert.py#L12) durch `return 'g0000000'` (oder irgendeinen anderen String), Bei mir hat's damit geklappt.

ralfsfb, 2023-03-17:
> habe selbst eine Entwicklungsumgebung, die ohne Internetverbindung auskommt. Dort benutze ich den zip Code. Ersetze mal in convert.py die [Zeile [...]

Würde den Eintrag ja gern ersetzen und damit neu testen, doch in dem ZIP-Archiv (https://github.com/lumapu/ahoy/archive/refs/heads/development03.zip) gibt es doch gar keine Datei, die convert.py heißt. Oder habe ich etwas übersehen?

beegee3, 2023-03-17:
ahoy-development03\src\web\html\convert.py (steht sogar in deinem Fehlerprotokoll oben direkt über der 'link = ...' Zeile 😄)

ralfsfb, 2023-03-17:
Jetzt hat's geklappt - Dankeschön!
Irgendwie war sowohl ich als auch die Dateisuche des W11-Explorers blind für die Datei.
Nach dem Download auf den ESP32 bleibt das Display bisher dunkel, doch das liegt vielleicht an einigen Änderungen, die seit meinem letzten Download vorgenommen wurden.

beegee3, 2023-03-17:
schön das es geklappt hat!
Display: vielleicht hilft die Diskussion in issue #759

ralfsfb, 2023-03-17:
Dank deines Hinweises funktioniert nun auch wieder das Display. Musste zwar auch das 1,3" Modell anstelle des vorhandenen 0,97" Modells konfigurieren, doch es hat auf jeden Fall geholfen.
Nochmals großes Dankeschön und ein sonniges Wochenende!

closed 2023-03-27 by lumapu