lumapu/ahoy #780: "MQTT is not connected" reason: "TCP disconnected

kriebsam, 2023-03-16:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ x] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [x ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [x ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Die AhoyDTU ist aktiv und erhält Daten vom Inverter. Inverter #0: Ba51-Inverter (v10018) is available and is producing; ABER es wird keine Verbindung zum MQTT-Broker aufgebaut. Daten für MQTT gesetzt; IP des Brokers, Port: 1833;
Im "Serial"-Monitor erscheint neben den Abfragewerten aus dem Inverter die Meldung: "MQTT disconnected, reason: TCP disconnect"

Im MQTT-Explorer taucht nur die IP von Home Assistant (.178.79) auf, nicht aber die IP der AhouDTU (.178.80)

Home Assistant und läuft auf einer VM meiner Synology 2018+

Meine Ideen sind mir ausgegangen. Kann jemand helfen?
Klaus

kriebsam, 2023-03-17:
Ich hatte vor ca. 2 Wochen erstmalig über den Online Installer (ahoydtu.de) den ESP programmiert und dann den Inverter (Hoymiles) in Betrieb genommen. Die Kommunikation hat aber erst funktioniert, nachdem ich die voreingestellte Wemos PIN-Belegung in der Ahoy-Konfig angepasst hatte. Eingestellt ist jetzt:
CS => D8(GPIO15)
CE => D3(GPIO0)
IRQ => D4(GPIO02)

Die MQTT hat auch nach vielen Neustarts der AHOYDTU noch nie eine Verbindung aufgebaut.

[image]

Beiliegend ein Auszug aus der: AHOYDTU (Serial/Control) sowie AhoyDTY (Systemdaten)
[AhoyDTU_Systemdaten.txt](https://github.com/lumapu/ahoy/files/10999589/AhoyDTU_Systemdaten.txt)

lumapu, 2023-03-23:
bitte mal die Developmentversion `0.5.105` testen, hier geht es um `0.5.66`

rmayergfx, 2023-03-29:
@kriebsam 
Bitte mal auf einem Rechner den MQTT Eplorer laufen lassen, aber nicht in der VM!
Ist es wirklich eine VM oder ein Docker Container? In welchem Netz läuft die VM? Wo läuft der MQTT Service?
Nutzt du den Mosquitto MQTT Broker? Ist in der Config anonymous erlaubt? 
So wie es sich anhört sind Ahoy DTU und VM, MQTT in unterschiedlichen Netzwerkbereichen und können sich deshalb nicht sehen.

kriebsam, 2023-03-29:
Ich hatte gestern die AhoyDTU auf die SW-Version: 0.6.0 upgedatet.Ergebnis:Nachdem ist den konfiguriereten User in der AhoyDTU eingetragen hatte wurde die Verbindung zum Mosquitto (der auf einer VM der Synology läuft) aufgebaut. Siehe beiliegender Screenshot.Mehr kann ich im Moment nicht sagen, da ich für eine Woche abwesed bin.Sieht aber erstmal gut aus.Vielen Dank für eure Arbeit!KlausVon meinem/meiner Galaxy gesendet

GallusMax, 2023-04-03:
> Es gibt bald eine neue MqTT Lib Verison, habe die Änderung schon erfolgreich getestet und löst bei mir das Problem:
> 
> [bertmelis/espMqttClient#86](https://github.com/bertmelis/espMqttClient/pull/86)
> 
> hier unsere Schritte zur Lösung: [bertmelis/espMqttClient#85](https://github.com/bertmelis/espMqttClient/issues/85)

MeToo: in der Version 0.6.0 hatte ich gestern einen Verbindungsabbruch MQTT "TCP disconnected". 
<ahoyIP/serial war weiter aktiv
<ahoyIP>/reboot brachte es wieder in die Spur

lumapu, 2023-04-04:
komisch, kommt das TCP disconnected immer wieder?

matze19999, 2023-07-18:
Ich hab das gleiche Problem. Ich hab noch überhaupt nicht geschafft eine MQTT Verbindung herzustellen. Im Log taucht die Fehlermeldung TCP disconnected alle paar Sekunden auf.

rmayergfx, 2023-07-19:
@matze19999 
Ohne Info mit welchem Broker und welchen Einstellungen du arbeitest wird es schwierig. Bitte mehr Infos. Ist anonymous aktiviert? Bitte mal einen Screenshot der AhoyDTU MQTT Settings hochladen. ServerIP kannst du gerne schwärzen.

matze19999, 2023-07-19:
> @matze19999 Ohne Info mit welchem Broker und welchen Einstellungen du arbeitest wird es schwierig. Bitte mehr Infos. Ist anonymous aktiviert? Bitte mal einen Screenshot der AhoyDTU MQTT Settings hochladen. ServerIP kannst du gerne schwärzen.

Sorry. Broker ist Mosquitto als Addon in Home Assistant installiert. Funktioniert aber auch nicht wenn Mosquitto direkt unter Linux installiert ist oder als Docker Container läuft.

Authentifizierung per Username / Password ohne Sonderzeichen.

Ahoy DTU EInstellungen:

[image]

Server IP fand ich sinnlos zu schwärzen bei einer privaten IP :)

lumapu, 2023-08-05:
das Problem sollte nicht exisitieren, bitte aktuelles Release `0.7.25` testen

closed 2023-09-16 by lumapu