lumapu/ahoy #781: Power-Limit Steuerung über mehrere WR hinweg

Pixel-Chris, 2023-03-17:
Hallo Zusammen,
ist es evtl. möglich eine Funktion einzubauen, die es ermöglicht ein Power Limit zu setzen, welches alle Inverter betrifft?

Sprich… man hat bspw. zwei WR angebunden (mit Ost/West Ausrichtung) und möchte nun fur beide WR ein gemeinsames PowerLimit von 600 Watt setzen. Hier müsste nun ein dynamisches PowerLimit greifen, welches beide WR dem entsprechend limitiert.

PixelChris

Ollipop030, 2023-03-17:
Auch in diesem Issue:

Sieh dir mal das an:

https://github.com/reserve85/HoymilesZeroExport

Macht genau das was du möchtest, ist aber ein Python Script und muss somit auf separater HW ausgeführt werden.

stefan123t, 2024-10-27:
@Knickohr war da nicht was geplant bzw. im Beta Test Stadium ?

knickohr, 2024-10-27:
„Bedarfsoptimierte Leistungsregelung“ die gerade in der heißen Testphase ist.