lumapu/ahoy #782: Nulleinspeisung geregelt von der AhoyDtu - Anbindung von externen PowerMetern

Pixel-Chris, 2023-03-17:
Hallo,
ist es möglich eine Funktion einzubauen, mit der eine Nulleinspeisung über die AhoyDTU realisiert werden kann.
Ohne Zuhilfenahme weitere Komponenten (MQTT Server, NodeRed usw)

Optimal wäre es bspw. wenn die so oder so durchlaufende AhoyDTU sich die Werte von einem PowerMeter (bspw. Shelly 3EM) abholt und mit diesen Werten dann das PowerLimit für die/den WR bestimmt.

PixelChris

Ollipop030, 2023-03-17:
Bisher nicht ohne separate HW, aber das Script macht das sehr gut:

https://github.com/reserve85/HoymilesZeroExport

Bei mir läuft das momentan auf einem Pi3. @lumapu kann man das so oder ähnlich in Ahoy integrieren?

dtuuser, 2023-03-17:
Bei mir läuft es über eigenes Tool. Arbeite gerade an einer 2 WR Logik....dauert aber noch. Eine Einbindung in Ahoy wäre klasse

dAjaY85, 2023-03-17:
Ahoy ist für die Kommunikation zum Hoymiles zuständig, das sollte auch so bleiben. Mann kann nicht jeden Fall abfangen, manche benutzen USB lesekopf, manche MQTT andere wiederum was ganz anderes wie victron oder so was.

Für Null Einspeisung benötigt man zusätzliche Hardware, mindestens einen anderen Controller.

homeautomation2022, 2023-03-18:
Einfach selbst einprogrammieren, wer es benötigt. Dann macht es auch genau das was man möchte.

bojs15, 2023-04-02:
hat jemand das mit raspberry pi pico W geschaft und plus shely3M ,oder gibt schon besere struktur

Pixel-Chris, 2023-04-06:
Vielleicht kann man hiervon profitieren:

https://github.com/helgeerbe/OpenDTU-OnBattery

tvellen, 2023-10-10:
Also DTU pro kann das auch meines Wissens.

stefan123t, 2023-12-21:
Das ist das selbe wie #781 nur jetzt eine Zero-Export Regelung in AhoyDTU nicht für mehrere Wechselrichter sondern die Anbindung der externen Power Meter als Eingang für die Regelgröße.

stefan123t, 2024-10-27:
M.E. ein Duplicate hier aber allgemein und in den beiden anderen Issues spezifischer, in #431 ModBus und #606 Tasmota / Shelly PowerMeter via HTTP(S)

ottelo9, 2025-06-11:
> Ja das wäre super wenn sowas gehen würde, hab auch ein HitchiIR Lesekopf am Stromzähler, wenn danach AhoyDTU den Wechselrichter drosseln könnte wäre Mega.

Genau das macht ja [OpenDTU-onBattery](https://github.com/hoylabs/OpenDTU-OnBattery)

Starfoxfs, 2025-06-11:
Auch AhoyDTU wie ich jetzt gelesen hab, es wird wohl nur der ESP32-S3 benötigt, dann gibt es die Firmware mit ZeroExport.

[https://docs.ahoydtu.de/de/latest/zeroExport.html](url)

Hab mir jetzt so nen ESP32-S3 bestellt die Dinger kosten ja nix.