lumapu/ahoy #784: Keine Kommunikation mehr obwohl alles gelaufen ist. Grund unklar

xxHighlanderxx, 2023-03-18:
## Hardware
  * [X ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X ] circuit board
* [ ] external antenna

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [X ] Windows
* [ ] Mac OS

Hallo!
Ich habe vor einigen Wochen mir die Hardware auf ESP8266 Basis gekauft (1x mit externer Antenne und 1x ohne).
Beide liefen einwandfrei. Bis vor ein paar Tagen. Ich hatte da gerade die 0.5.92 (oder eine Version danach) drauf und dacht zuerst es ist bestimmt ein Bug und wird zeitnah wieder verschwinden. Leider war das nicht der Fall. Die Kommunikation fiel dann komplett aus und ist es in 0.5.102 immer noch.
Ich habe dann auf der anderen Hardware die 0.5.66 installiert und den Wechselrichter eingerichtet. Auch dort gibt es keine Kommunikation mehr. Nun  dachte ich es liegt am WR und habe diesen heute gegen einen baugleichen neuen getauscht. SN angepaßt und wieder keine Kommunikation. Ich bin nun komplett ratlos.
Wie kann ich der Sache nun auf den Grund gehen. Die Eintragungen im Logbuch sagen mir nichts.

Unter Serials / Control sehe ich folgendes:

0.5.66
13:09:56 I: resetPayload: id: 1
13:09:56 I: (#1) sendTimePacket
13:09:56 I: sendTimePacket b
13:09:56 I: TX 27B Ch40 | 15 83 30 32 18 84 31 11 77 80 0b 00 64 15 aa 14 00 00 00 00 00 00 00 00 2b 4c 7c 
13:10:31 I: resetPayload: id: 1
13:10:31 I: (#1) sendTimePacket
13:10:31 I: sendTimePacket 5
13:10:31 I: TX 27B Ch61 | 15 83 30 32 18 84 31 11 77 80 05 00 64 15 aa 37 00 00 00 00 00 00 00 00 d4 ce 2c 
13:11:06 I: resetPayload: id: 1
13:11:06 I: (#1) enqueuedCmd: 1
13:11:06 I: (#1) enqueuedCmd: 11
13:11:06 I: (#1) enqueuedCmd: 5
13:11:06 I: (#1) sendTimePacket
13:11:06 I: sendTimePacket 1
13:11:06 I: TX 27B Ch75 | 15 83 30 32 18 84 31 11 77 80 01 00 64 15 aa 5a 00 00 00 00 00 00 00 00 82 3b e6 

0.5.102
I: resetPayload: id: 0
I: (#0) prepareDevInformCmd 0xb
I: TX 27B Ch3 | 15 83 30 32 18 89 70 04 87 80 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 ca a1 16 
I: resetPayload: id: 0
I: (#0) enqueuedCmd: 0x1
I: (#0) enqueuedCmd: 0xb
I: (#0) prepareDevInformCmd 0x1
I: TX 27B Ch23 | 15 83 30 32 18 89 70 04 87 80 01 00 00 00 00 00 00 00 00 00 00 00 00 00 c0 ab 1c 
I: resetPayload: id: 0
I: (#0) prepareDevInformCmd 0xb
I: TX 27B Ch40 | 15 83 30 32 18 89 70 04 87 80 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 ca a1 16 

Ich habe die Geräte auch direkt neben dem Wechselrichter gehabt, aber keine Änderung.

Wie kann ich der Sache weiter auf den Grund gehen?

Ich bräuchte Eure Hilfe.

Danke und Gruß

Christian

Argafal, 2023-03-18:
Die TX-Zeilen im Log ist deine DTU (also Ahoy) die Pakete an den Wechselrichter schickt um nach Daten zu fragen. Es kommt keine Antwort, das wären RX Zeilen. Wenn das schon mal funktioniert hat dann ist das natürlich seltsam. 

Hab ich richtig verstanden, dass du zwei verschiedene ESPs probiert hast aber das gleiche NRF-Funkmodul mit beiden? Vielleicht ist das Funkmodul kaputt gegangen? Oder eine Pin-Einstellung ist beim Hin- und Herbauen nicht richtig übernommen worden? Weder bei 0.5.66 noch bei 0.5.92 noch bei 0.5.102 hab ich Berichte von Kommunikationsschwierigkeiten gesehen, die mit deinen Symptomen zusammenpassen würden. Hm. Most curious.... Ich drück erst mal die Daumen, vielleicht hat ja jemand anderes noch eine Idee. Wenn's mein Setup wäre würde ich mal ein anderes NRF-Modul probieren.

xxHighlanderxx, 2023-03-18:
Hallo Argafal! Vielen Dank für die Infos. Ich habe die Hardware nicht umgebaut. Es sind zwei getrennte Platinen. Eine ist in einer Kiste und die Platinen sind zusammengesteckt. Das andere sind die Platinen mit Kabel dazwischen.
Gelaufen sind beide nun seit mindesten 3-4 Monaten ohne nennenswerte Vorkommnisse.
Es stellt sich nun die Frage warum kann ich nichts mehr empfangen? Gibt es ggf. eine Störung die durch etwas anderes verursacht worden sein kann? Also ich meine eine Art Trägerfrequenz die stört? Klingt aber auch abwegig, da ich ja auch direkt daneben nichts bekomme. Ich bin daher ratlos. 2 WR, 2 komplett Module zum Empfangen - alles fällt sozusagen zur gleichen Zeit aus? Leider bleibt es weiter mysteriös.

tqma1, 2023-03-18:
Hi @xxHighlanderxx !

Hast Du das PinOut in den Ahoy Settings überprüft?
Hatte ich schon 2-3 Mal das dieses plötzlich nicht mehr passte.

xxHighlanderxx, 2023-03-18:
Ja, eigentlich habe ich es geprüft. Beide sind gleich.
[image]

Argafal, 2023-03-18:
Wenn beide gleich sind, stimmt für deinen zweiten ESP die Konfiguration nicht. Du hast weiter oben geschrieben:

[image]

Das stimmt mit dem Konfigurations-Screenshot nicht überein. Vielleicht kannst du versuchen, die Software-Einstellungen so anzupassen, dass sie der Hardware entsprechen?

xxHighlanderxx, 2023-03-19:
OK, ich habe aber den 8266. Der wäre doch wie mein Screenshot oder übersehe ich jetzt etwas?

xxHighlanderxx, 2023-03-19:
Es ist ein wenig komisch:
Ich habe seit gestern Abend nicht gemacht, aber die 0.5.66 kommuniziert nun wie durch Geisterhand ( ich habe die SN am Ende mit UVWX ersetzt):

07:41:04 I: procPyld: cmd:  11
07:41:04 I: procPyld: txid: 0x95
07:41:04 I: Payload (62): 00 01 01 48 00 2f 00 2d 00 99 00 94 00 00 05 ff 00 00 06 16 00 06 00 06 01 49 00 2f 00 2d 00 9a 00 93 00 00 06 03 00 00 05 f2 00 06 00 06 09 0b 13 89 02 3d 00 d1 00 19 03 ac 00 89 00 09 
07:41:33 I: resetPayload: id: 0
07:41:33 I: (#0) Requesting Inv SN 11618330UVWX
07:41:33 I: (#0) enqueuedCmd: 11
07:41:33 I: (#0) sendTimePacket
07:41:33 I: sendTimePacket b
07:41:33 I: TX 27B Ch3 | 15 83 30 32 18 84 31 11 77 80 0b 00 64 16 ae 9d 00 00 00 09 00 00 00 00 2d d5 64 
07:41:33 I: RX 27B Ch61 | 95 83 30 32 18 83 30 32 18 01 00 01 01 4b 00 2f 00 2d 00 9b 00 96 00 00 05 ff 2a 
07:41:33 I: RX 27B Ch40 | 95 83 30 32 18 83 30 32 18 02 00 00 06 16 00 06 00 06 01 4c 00 2f 00 2d 00 9b 53 
07:41:33 I: RX 27B Ch40 | 95 83 30 32 18 83 30 32 18 03 00 95 00 00 06 04 00 00 05 f2 00 07 00 06 09 08 f6 
07:41:33 I: RX 27B Ch23 | 95 83 30 32 18 83 30 32 18 84 13 8a 02 42 00 d1 00 19 03 ad 00 89 00 09 86 a0 08 
07:41:34 I: procPyld: cmd:  11
07:41:34 I: procPyld: txid: 0x95
07:41:34 I: Payload (62): 00 01 01 4b 00 2f 00 2d 00 9b 00 96 00 00 05 ff 00 00 06 16 00 06 00 06 01 4c 00 2f 00 2d 00 9b 00 95 00 00 06 04 00 00 05 f2 00 07 00 06 09 08 13 8a 02 42 00 d1 00 19 03 ad 00 89 00 09 

Die 0.5.102 macht immer noch nichts:

07:42:35 I: TX 27B Ch40 | 15 83 30 32 18 89 70 04 87 80 0b 00 64 16 ae db 00 00 00 00 00 00 00 00 4f 13 26 
07:43:05 I: enqueued cmd failed/timeout
07:43:05 I: (#0) no Payload received! (retransmits: 0)
07:43:05 I: resetPayload: id: 0
07:43:05 I: (#0) Requesting Inv SN 11618330UVWX
07:43:05 I: (#0) enqueuedCmd: 0x1
07:43:05 I: (#0) enqueuedCmd: 0xb
07:43:05 I: (#0) prepareDevInformCmd 0x1

Ist doch spannend, oder? Leider fehlt mir hier immer noch das Verständnis was nun anders ist. Die Einstellungen habe ich nun 3x miteinander verglichen und die sind gleich. 

Kann es ein Problem sein, dass der WR von 2 Devices abgefragt wird?

tqma1, 2023-03-19:
Mehrere DTUs die einen Wechselrichter Abfragen sind grundsätzlich kein Problem es dauert einfach länger bis jede DTU gültige Antworten erhält.

Zu diesem tatsächlichen seltsamen Verhalten kann ich nichts sagen außer das ich mal einen Wechselrichter hatte der erst mit dem 3. NRF-Funkmodul wollte, andere WR hatten mit den anderen Modulen kein Problem.

xxHighlanderxx, 2023-03-19:
OK. Danke. Beide Module haben ja schon mal über Wochen funktioniert. Ich habe auch an der Hardware der .102 nichts geändert. Die ist in einer geschlossenen Box die ich noch nie geöffnet habe (hatte ich so erworben und hat ja einwandfrei funktioniert).

dtuuser, 2023-03-19:
Ich frage seit Monaten meinen WR mit 2 DTU's ab. DTU 1 alle 15 Sekunden, DTU 2 alle 63 Sekunden.
DTU 1 regelt die Nulleinspeisung (aktuell mit 5.92) und DTU 2 bekommt jedes Update umgehend drauf.

@xxHighlanderxx schau mal bitte nach, ob das NRF Modul an D3 und D4 angeschlossen ist. Mein erster Wemos läuft problemlos mit D3 und D4. Meine Test DTU  habe ich nur mit D1 und D2 ans Laufen bekommen. Ist irgendwas mit IRQ  an D4 und zu nah an Ground.
Es könnte auch am Netzteil und oder am 100er Kondensator liegen.

xxHighlanderxx, 2023-03-19:
Vielen Dank für die Tips. Ich kann an dem einen nichts ändern, weil die beiden Platinen mit einer "Adapterplatine" verbunden sind. 
Netzteil werde ich wechseln und es weiter beobachten. Sollte sich die Lösung noch ergeben, dann teile ich natürlich das "Geheimnis" mit Euch.

xxHighlanderxx, 2023-03-20:
Ich habe nun nochmal die zweite Hardware komplett zurückgesetzt und die 0.5.66 installiert (über den automatischen Installer auf ahoydtu). Die Kommunikation steht nun wieder. D.h. es muss wohl etwas in der Beta-Version ab ca. .9x sein, was bei meinen Geräten die Kommunikation verhindert. Gerne können wir der Sache weiter auf den Grund gehen, aber da bräuchte ich jemanden von Euch der sozusagen die Tests begleitet. Ich habe nun als ersten Test ein Update auf .104 gemacht. Was soll ich sagen. Keine Kommunikation mehr. Spannend!

xxHighlanderxx, 2023-03-20:
Ja. Danke, aber es betrifft auch die verkabelte Variante.

Argafal, 2023-03-20:
Ich muss gestehen, ich versteh dein Setup nicht wirklich. Das klingt für mich ohne es je gesehen zu haben alles so kompliziert. Doppelte Hardware, keine Ahnung was wirklich verkabelt ist.... 

Wenn's mein Setup wäre würde ich die Komplexität verringern: Nur einen Satz Hardware. Genau die Verkabelung anschauen, durchmessen, aufschreiben/aufmalen. Dann die Konfiguration kontrollieren ob das auch wirklich so den Einstellungen entspricht. Fotos machen. Das Problem eingrenzen. Ansonsten ist das alles nur ein Ratespiel, da bin ich nicht so gut drin... :)

Ollipop030, 2023-03-20:
Es gab hier auch mal ein Issue eines anderen Anwenders, der hatte eine DTU von einem Ebay Händler gekauft. Die lief anscheinend out of the Box, aber nach einem Update ging das Teil auch nicht mehr.

lumapu, 2023-03-25:
@xxHighlanderxx gibt es schon ein Update?

xxHighlanderxx, 2023-03-25:
Guten Morgen! Ich habe das Pinout kontrolliert und es entspricht dem "Standard" gemäß Schaltungsbild anbei. Ansonsten noch ein paar Fotos zur Illustration. Leider bin ich nun ratlos und werde sozusagen kapitulieren - ggf. hat ja noch jemand eine Idee, aber eigentlich sehe ich das Projekt aktuell in Richtung Abschluß - sprich warten auf den nächsten Release.

[image]
[image]
[image]
[image]

lumapu, 2023-03-25:
die Kabel sind schon sehr lang!
Kannst du bei deinen NRF prüfen, ob es Lötfehler aus der Produktion gibt, das hatte ich bei einem meiner Module

dtuuser, 2023-03-25:
100er Kondensator besser direkt an die NRF Platine, oder?

xxHighlanderxx, 2023-03-25:
Vielen Dank für die Hinweise. Ich werde den nächsten Test mit dem nächsten Release machen. Nachdem ich wohl der einzige bin ist die Wahrscheinlichkeit hoch, dass es an der Hardware liegen könnte. Bei dem anderen kann ich ja leider nicht reinschauen, aber es ist schon komisch, dass zwei verschiedenen Hardwarekomponenten sozusagen "zusammen" nicht mehr laufen. Wie auch immer, vielen Dank für Eure Hilfe. Ich denke Du kannst die Diskussion hier schließen.

closed 2023-03-25 by xxHighlanderxx

lumapu, 2023-03-25:
ja genau der Kondensator muss zum NRF, der Einfachheit halber und wegen den langen Kabeln einfach einen zweiten einbauen

xxHighlanderxx, 2023-03-31:
Nachtrag: Ich habe nun die 0.6.0 drauf, nichts an der Hardware verändert. Was soll ich sagen: läuft. Also scheint es ja etwas in der Entwicklungsversion gewesen zu sein. Dies zur Info.