lumapu/ahoy #786: Steuerung über MQTT

PeterObs, 2023-03-18:
Ich nutze einen 8266 mit Version 0.5.77 für eine Nulleinspeisung. Nach Update auf .102 empfängt die DTU keine MQTT Pakete. Der Zähler für empfangene Pakete bleibt bei Null und es erfolgt keine Steuerung des PowerLimit,  Pakete senden  funktioniert ( kann ich auch empfangen). Nach einem Downgrade auf .77 geht alles wieder. 

Noch eine Frage: die Bestätigung power_limit_ack bedeutet das die DTU das Limit empfangen hat oder der Inverter?

B4unty, 2023-03-18:
Das selbe Problem habe ich auch mit inverter/WL1/devcontrol/0/11/1 den inverter angesprochen hat sich da was verändert?

dtuuser, 2023-03-19:
Da hat sich seit der Version 5.98 was geändert.

https://github.com/lumapu/ahoy/blob/main/User_Manual.md

Power Limit absolute non persistent [Watts]
<TOPIC>/ctrl/limit_nonpersistent_absolute/<INVERTER_ID>             with a payload [0 .. 65535]

Example:         inverter/ctrl/limit_nonpersistent_absolute/0     600

Ich meine aber, dass hinter der 600 noch ein W sein muss.

beegee3, 2023-03-19:
falscher Hinweis, das ist  noch das alte Manual der letzten release Version 5.66
Das dev. Manual ist https://github.com/lumapu/ahoy/blob/development03/User_Manual.md

**Power Limit relative (non persistent) [%]**
<TOPIC>/ctrl/limit/<INVERTER_ID>
with a payload [2 .. 100]

Example:

inverter/ctrl/limit/0     70

**Power Limit absolute (non persistent) [Watts]**
<TOPIC>/ctrl/limit/<INVERTER_ID>
with a payload [0 .. 65535]

Example:

inverter/ctrl/limit/0     600W

PeterObs, 2023-03-19:
Danke, hatte noch das alte Manual. Auch die Information über die Funktion der Topic ack_pwr_limit ist damit geklärt.

closed 2023-03-19 by PeterObs