lumapu/ahoy #787: ESP32 Kommunikation mit NRF24L01+ Modul

KoLSMS, 2023-03-19:
Hallo,
habe heute ein ESP32 Board mit einem NRF24L01+ Modul verdrahtet, leider zeigt mir die Weboberfläche der AhoyDTU die Meldung an nrf24l01+ is not connected.
Habe ich bei der Verdrahtung einen Fehler gemacht?!
Ich hoffe man kann auf den Bildern alles gut erkennen 🙂
Gruß...

[image]
[image]
[image]
[image]
[image]

teletapi, 2023-03-19:
Die Bilder sind in teilen so groß und unscharf und noch gezoomt das man garnix mehr richtig erkennt

KoLSMS, 2023-03-19:
> Die Bilder sind in teilen so groß und unscharf und noch gezoomt das man garnix mehr richtig erkennt

Hallo,

so sieht man es besser?!

Gruß...

[image]
[image]
[image]
[image]

Argafal, 2023-03-19:
Hast du noch einen Screenshot von "Einstellungen" "System Config"? Da ist das Pinout definitert.

KoLSMS, 2023-03-19:
> Hast du noch einen Screenshot von "Einstellungen" "System Config"? Da ist das Pinout definitert.

Hallo, ja, selbstverständlich:

[image]

Argafal, 2023-03-19:
Sieht in meinen Augen gut aus. Vielleicht ist der NRF kaputt? oder eine nicht-plus Version? Der Kondensator falsch rum? Die Verdrahtung sieht so aus, wie ich sie erwartet hätte. 
Irgendwo gab es hier einen anderen Issue da hatte jemand Fotos von zwei NRF Modulen gepostet. Beide sahen gleich aus, eins ging, das andere nicht...

Tofffii, 2023-03-21:
Die Verdrahtung sieht auch für mich gut aus. Die Lötpunkte am Kondensator wirken recht groß, sind auf den Fotos aber nicht richtig sichtbar, ist ein Kurzschluss sicher ausgeschlossen?

KoLSMS, 2023-03-22:
Hallo,

vielen Dank für die Antworten, ich habe den Fehler nun endlich gefunden...

Es war alles richtig, der ESP32 Chip hat einen Schuss weg...

Neuer ESP32 und schon läuft alles...

Ärgerlich wenn man voraussetzt das der Chip in Ordnung ist...

Vielen Dank nochmals...

PS: Woran ich das gemerkt habe?

Ich habe die Verbindung zwischen ESP32 und NRF getestet mit einem Messgerät, dabei ist mir per Zufall aufgefallen das GND und GPIO23 eine Verbindung haben, was sowas von keinen Sinn machte das ich das am anderen Chip auch getestet habe, hier Bestand keinerlei Verbindung und schon ging alles...

Tja, der Fehler liegt im Detail :)

closed 2023-03-22 by KoLSMS