lumapu/ahoy #789: Ist es möglich Zählerstände im Wechselrichter zu setzen?

einencool, 2023-03-19:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: Wemos D1 Mini______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.66
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [x] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo zusammen,
nach kurzer Suche konnte ich leider keine Informationen dazu finden, daher frage ich hier mal nach.

Ich habe einen Hoymiles HM-600, Kommunikation und alles funktioniert TOP, jetzt soll dieser aber bei mir durch einen HM-800 ersetzt werden und der jetzige HM-600 wird in der Verwandschaft weiter verwendet.

Ist es möglich, die "YieldTotal" Werte zu löschen, bzw meine alten Werte in den HM-800er einzugeben?
So würden meine Werte in der Hausautomation nahtlos ineinandergreifen.

Über eine Rückmeldung würde ich mich sehr freuen.
Gruß
Chris

dtuuser, 2023-03-19:
Geht unter Settings-Inverter-Yield Total Correction.
Einfach den aktuellen Wert vom WR bei der Konfiguration vom neuen WR dort ein tragen.
Geht aber erst ab Version 5.86 glaube ich.
Meine 5.92 läuft jetzt seit 14 Tagen und 6 Stunden super stabil

einencool, 2023-03-19:
Oh perfekt,
Dann hätte ich doch mal vorher ein Update machen sollen :-)

Vielen Dank für die schnelle Rückmeldung 👍

closed 2023-03-19 by einencool