lumapu/ahoy #791: kein Display Setting Menü in 0.5.66?! ESP32 SSD1306 + SSD1306/1106 an 8266

grericht, 2023-03-19:
Hallo. Bin erst neu bei Github angemeldet und hoffe nicht gleich einen Fehler durch eine zu Dumme Frage zu begehen.

Ich bastel gerade etwas an einer ahoyDTU herum. Bin bisher super zufrieden und wollte nun um ein Display erweitern. Da ich 2 Freunde mit versorge hab ich mir fast alle Kombinationen bei Ali.*.* bestellt.

Ich habe bisher mit dem Webinstaller die letzte stable installiert. 0.5.66 müsste das sein. Danach bleibt mein Display am ESP32 dunkel. Der rest geht. Wenn ich dann die Datei 221230_ahoy_0.5.66_esp32_ssd1306_f8fe044.bin per web-gui update installiere. geht das Display an. Aber ich habe kein Menü Display Settings. Gibt es das ncoh nicht in der Rev?

Wie bekomme ich eine Dev-Rev. installiert und welche Version ist da empfohlen? Die neueste? Hab nur die 230319_ahoy_0.5.103_296597a_esp32.bin  per web-gui update installiert aber dabei "fährt der esp32 fest". Muss dann wieder komplett auf die 0.5.66 zurück udn die display.bin updaten. Entschuldigt meine dummen fragen aber ich finde wirklich kaum etwas zum Thema Display anschließen.

die nächste Frage wäre: ich hab mir gleich noch das 1306 und das halb so große 1306 mitbestellt. Die würde ich gerne als nächstes anschließen. Dafür gibt es aber bei der 0.5.66 keine update Datei.

Last but not least: zum 8266 finde ich gar keien Darstellung zur Verkabelung mit den Displays. 5v udn gnd finde ich sicher. Aber woran kommn die beiden Signal-Leitungen?

Ich danke euch!

mroenne2022, 2023-03-20:
[image]

Das wird so zumindest in der FB Gruppe kommuniziert, ich selbst hab aber kein Display.

knickohr, 2023-03-20:
Das ist die offizielle Verdrahtung für eine Ahoy-DTU mit OLED

[image]

Ein Display-Setting findest Du für die Version 0.5.66 noch nicht, einfach nur das richtige Binary auswählen und gut ist,

Das größere 1,3“ SH1106 wird mit der 66 auch noch nicht supportet. Nimm eine Developer-Version.

Developer-Versionen gibt es hier :
https://nightly.link/lumapu/ahoy/workflows/compile_development/development03/ahoydtu_dev.zip

grericht, 2023-03-20:
vielen vielen Dank! Super Community hier!
Hab jetzt auch die aktuellste dev installiert bekommen. Hatte den Hinweis "Bekommt man nach der Auswahl des COM-Ports einen Fehler Failed to download manifest muss man Chrome mit einem Parameter starten: Windows start chrome --allow-file-access-from-files" übersehen. 
Am 8266 werde ich die displays zeitnah testen.

Ich habe auf dem Display links oben aber ein paar fehlende Pixel. Ist das normal? Muss mal ein anderes Display anschließen und schauen. Ist genau dort wo die Aktuelle Leistung angezeigt wird. also ein no-Go

Ich hab vorher von der 0.5.66 eine jsan sicherung gemacht und hinterher eingespielt. Dabei hat es aber keien Daten übernommen? Liegt das daran, weil die Versionen (zu) unterschiedlich sind?

Perspektivisch würde ich gerne versuchen selber ein wenig zu schauen wie/wo ich etwas auf dem Display anzeige. Gibt es dafür eine gute Beschreibung wie ich mich damit gut auseinandersetzen kann?

Argafal, 2023-03-20:
> vielen vielen Dank! Super Community hier! Hab jetzt auch die aktuellste dev installiert bekommen. Hatte den Hinweis "Bekommt man nach der Auswahl des COM-Ports einen Fehler Failed to download manifest muss man Chrome mit einem Parameter starten: Windows start chrome --allow-file-access-from-files" übersehen. Am 8266 werde ich die displays zeitnah testen.
> 
> _Ich habe auf dem Display links oben aber ein paar fehlende Pixel. Ist das normal? Muss mal ein anderes Display anschließen und schauen. Ist genau dort wo die Aktuelle Leistung angezeigt wird. also ein no-Go_ EDIT: hat sich erledigt. Keine AHnung woran es lag aber nachdem ich es neu angeschlossen hab waren alle Pixel da
> 
> Frage 01: Ich hab vorher von der 0.5.66 eine jsan sicherung gemacht und hinterher eingespielt. Dabei hat es aber keien Daten übernommen? Liegt das daran, weil die Versionen (zu) unterschiedlich sind?
> 
> Frage 02: Perspektivisch würde ich gerne versuchen selber ein wenig zu schauen wie/wo ich etwas auf dem Display anzeige. Gibt es dafür eine gute Beschreibung wie ich mich damit gut auseinandersetzen kann?

Es hat sich viel seit 0.5.66 einiges geändert. Ich denke, neue Einstellungen setzen ist die beste Lösung. Macht man ja auch nur einmal.

Loetnase, 2023-03-21:
Hallo @knickohr ,
liegen die fritzings mit Display (auch Nokia neuester Stand) irgendwo ab?
In Discord habe ich die gesehen, waren aber nicht runterladbar und etwas unscharf.

Loetnase, 2023-03-24:
Die Auflösung sollte hoch sein damit beim ESP32 möglichst die GPIO Beschriftung und auch sonstige Beschritungen noch lesbar sind. Wenn das möglich ist? Danke

knickohr, 2023-03-24:
[image]
[image]
[image]
[image]
[image]
[image]

filo4444, 2023-07-18:
Please add descriptions of capacitors capacity.

Loetnase, 2023-08-11:
@Silece87 
100uF 5V oder mehr. Der Kondensator am Antennenmodul nrf24 ist Pflicht, die zwei andern "nice to have".
Meist wird der Kondensator direkt am Antennenmodul an den +3,3V und GND Pins oben angelötet. Dabei auch die Polarität des Kondensators beachten, meist ist ein - an dem Anschlusspin am Gehäuse aufgedruckt, der kommt auf GND.

Silece87, 2023-08-14:
Hi,
Vielen Dank für die Info, dass 5V steht für 5 Volt? Könnte ich auch einen 100uF mit 35 V verwenden? Oder meinst du mit mehr, mehr farad.
Viele Grüße

Loetnase, 2023-08-14:
@Silece87 
das bezog sich auf die 5 Volt, wenn die eine größere Spannung z.B. 35 Volt können funktionieren die auch mit 5 Volt sind aber mechanisch größer. 
Mehr uF ist auch okay aber auch mechanisch größer.

Loetnase, 2023-08-14:
@Silece87 
Nein, ohne Kondensatoren passiert dem Board nichts, das sind in dem Fall nur Pufferspeicher, die das Ganze stabiler funktionieren lassen.  Das funktioniert oft auch ohne die Kondensatoren, aber bei vielen komischen Problemen sind die mit Kondensatoren plötzlich weg. Suche mal nach den 10 Geboten von Knickohr.

Silece87, 2023-08-14:
Ich habe die Anschlüsse 10mal geprüft und auch diverse Leute drüber schauen lassen.Wenn ich es ohne Kondensatoren an 5volt anschließe wird es Sau heiß und kurz danach fängt es an zu qualmen. Kein Grund /plus Verwechslung,alles probeweise getauscht.

knickohr, 2023-08-15:
Also wenn’s heiß wird und qualmt, ist auf alle Fälle schon mal was kaputt.

Ab in den Müll !

Loetnase, 2023-08-15:
Ja da hat Knickohr wieder mal zu 100% recht.
Aber was wird denn da heiss und qualmt?
Versuche das mal schrittweise zu testen.
1. Schritt: Mach mal alle Leitungen vom ESP weg und stecke das USB Kabel an einem USB Netzteil ein. Wird der ESP heiss bzw qualmt, dann ist der defekt.
2. Schritt: USB Kabel am PC einstecken, schauen ob sich der ESP am USB meldet und sich eventuell flashen lässt.
3. Schritt: Versuche den ESP ganz normal als AhoyDTU ins WLAN zu nehmen. Hat das funktioniert ist der ESP i.O.
Natürlich kann er dabei die Antenne nicht finden und ist nicht connectet.
4. Schritt: USB ist ausgesteckt. Verbinde nur die schwarze Leitung GND und rote Leitung 3,3V vom ESP zum Antennemodul, jetzt USB einstecken und beobachten ob und was was heiss wird. Vieleicht machst du das ohne Kondensator, der könnte falsch gepolt sein. Versuche den ESP als AhoyDTU im WLAN zu finden. Wenn das geht ist bis jetzt alles i.O. Wenn nein Antennenmodul defekt.
5. Schritt: Kondensator richtig rum anlöten. Was ist nun i.O. oder n.i.O. (lag der Fehler an dir oder am Kondensator)
6. Schritt: die restlichen 6 Leitungen verbinden und schauen ob jetzt  i.O. oder n.i.O.

Vieleicht auch hier mal ein Bild vom angelöteten Kondensator posten, das - (minus) am Kondensator sollte zu sehen sein.
Viel Erfolg

knickohr, 2023-08-15:
Punkt 1 :  Wenn’s mal qualmt isses IMMER kaputt. Es qualmt meißt such nur einmal 💥

 Punkt 4 und 5 :  Wenn der Kondensaor mal falsch herum war ist er entweder schon explodiert 💥 oder aber zumindest so arg angeschlagen das ich ihn nicht mehr verwenden würde. Denn über kurz oder lang mach er sich von dannen und reißt womöglich noch was (neues) mit.

Loetnase, 2023-08-15:
Fehler zu suchen, finden und verstehen ist immer lehrreich.
Denn "aus Fehlern wird man klug, drum ist einer nicht genug".
Aus dem Schrott kann auf jeden Fall noch etwas gelernt werden.
Ich denke, du hast da wie ich auch schon viel gelernt :joy:
Du hast aber recht, angerauchte und "madige" Teile ghören erneuert.