lumapu/ahoy #794: [ESP8266] MQTT Home Assistant Auto Discovery mit 2 Invertern klappt nicht

ottelo9, 2023-03-21:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection diagram

## Connection diagram I used:
-

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.104

### Github Hash

25695fc

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: -
    - Name: HM800-Schuppen
#### Inverter 1
    - Address: -
    - Name: HM800-Carport
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: meine Home Assistant IP
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: HM800

### Error description

Ich verwende Home Assistant 2023.3.5 mit Mosquitto broker 6.1.3 und der Integration Mosquitto broker. Bisher habe ich mit Ahoy einen Inverter HM-800 ausgelesen und via MQTT an HA gesendet ohne Probleme! Nun habe ich einen zweiten HM-800 eingebunden. Beide werden von Ahoy gefunden und korrekt ausgelesen. Nur leider wird der zweite nie in dem Moquitto Broker dargestellt. Daraufhin habe ich Ahoy auf Werkseinstellung zurück gesetzt und alles neu eingegeben ... ohne Erfolg. Die Broker Integration in HA gelöscht, HA neugestartet und dann neu hinzugefügt -> findet Ahoy nun garnicht mehr automatisch. Erst wenn ich auf den Ahoy Button "Discovery Config (homeassistant)" klicke, wird der erste Inverter in HA gefunden. Mit dem inoffiziellen Addon MQTT Explorer in HA werden aber beide Inverter angezeigt. Scheint also ein Problem mit dem Auto Discover zu geben?

[image]

ottelo9, 2023-03-21:
Ein anderer Nutzer aus Discord hat auch bestätigt, dass die Auto Discovery Message nur den ersten Inverter beinhaltet.

ottelo9, 2023-03-21:
Aktuell verwende ich den Workaround aus dem Discord Chat Kanal #home-assistant. Da muss man zum Anfang gehen und die dort veröffentlichte yaml config Datei in config\packages\ einfügen und anpassen. Dann hat man beide Inverter manuell hinzugefügt.

lumapu, 2023-03-23:
finde eigenartig, dass es wieder 'kaputt' sein soll. Ein Inverter geht, ab dem zweiten hakts?

ottelo9, 2023-03-24:
In welcher Version ging es denn mal? Die würde ich ja dann nehmen :). Ich habe auch mal testweise alle Inverter aus Ahoy rausgeschmissen und dann neu hinzugefügt. Der Fehler bleibt aber. Nur der erste Inverter wird in HA angezeigt.

lumapu, 2023-03-24:
Habe gerade das changelog nochmal gelesen, daher an Version 0.5.79.
Sollte sich meiner Meinung nach aber auch nicht wieder verschlechtert haben.

https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md

ottelo9, 2023-03-24:
Kann gut sein. Ich kann ja mal "irgendwie" die Discovery Message abhorchen, wenn ich in Ahoy auf den Button klicken. Wie kann man das am besten machen? 

- Mosquitto broker ?
- MQTT Explorer?
- ?

KG3RK3N, 2023-03-24:
@ottelo9 
Ich nutze dafür in der Regel [MQTT Explorer](http://mqtt-explorer.com/)
Das müsste dann in etwa so aussehen damit es von HA erkannt wird:
[image]
[image]

Wichtig ist, das unter dem Topic "homeassistant" die Config gepublished wird wie in den beiden Bildern zu sehen, damit HA das dann auch erkennt. Dort sind dann die Konfigurationen der einzelnen Sensoren enthalten, damit HA dann weiß worauf er horchen muss um Werte zu bekommen. 
In deinem Bild aus dem initialen Kommentar sind nur die Werte von ahoy selbst zu sehen aber nicht die für HA

closed 2023-03-27 by lumapu