lumapu/ahoy #796: Keine Verbindung zum Wechselrichter

pade1984, 2023-03-22:
Hallo,

ich habe einen HM-600 und möchte diesen mit dem Ahoy DTU verbinden. Als Aufbau habe ich einen D1 mini (ESP8266) und das NRF24L01+ Modul. Angeschlossen habe ich wie folgt:

**ESP     NRF24L01+**
G        GND
D3      IRQ 
D4      CE
D5      SCK
D6      MISO
D7      MOSI
D8      CSN

Im Ahoy DTU Menü habe ich folgende Einstellungen übernommen:
[image]

Ich bekomme jedoch keine Daten vom Wechselrichter (Logs s.u.)
[image]

Die Inverter-Seriennummer habe ich geprüft. Das ist die, die auf der Rückseite vom HM-600 steht. Habt ihr eine Idee woran das liegen könnte?

Viele Grüße

Philipp

Argafal, 2023-03-22:
Mach doch mal ein paar gute Fotos von deinem Aufbau (Übersicht und Detail), vielleicht sieht ja jemand was.

Ich würde an deiner Stelle auch mal probieren, statt D3 einen anderen Pin zu verwenden. D1 zum Beispiel (Hard- und Software, beides muss angepasst werden). Hilft das? Dann kannst du noch probieren statt D4 einen anderen Pin zu verwenden, D2 zum Beispiel.

pade1984, 2023-03-22:
Andere PINs habe ich schon versucht (Schritt für Schritt), aber ohne Erfolg.

Anbei ein paar Bilder von meinem Aufbau.
[image]
[image]
[image]
[image]

dtuuser, 2023-03-22:
Auf was hast du denn unter der Konfig das Amplifier Power Level stehen? Habe bei mir das Problem, dass es nur stabil mit LOW läuft. Das mit Version 5.66 jetzt schon seit 17 Tagen und 4 Stunden.

pade1984, 2023-03-22:
Ich hatte es zuvor auf High, aber auch auf LOW oder MIN ändert sich nichts :-(

dtuuser, 2023-03-22:
Komisch. 
Ich musste bei zwei von drei Boards D1 und D2 nutzen. Jeweils zweimal neu starten, da beim ersten Mal die Änderungen nicht übernommen wurden.

pade1984, 2023-03-22:
Auch schon mehrfach probiert. Funktioniert (leider) nicht.

mtet2, 2023-03-23:
Meine Boards laufen nur mit D1 und D2.
Bei allen HM-600, die mir bisher in die Finger gekommen sind, fängt die SNR mit 1141.... an.
Und die HMs antworten nur wenn > 1,1 W PV Leistung anliegt. Aber das sollte ja bekannt sein.
Besteht die Möglichkeit an einem anderem HM zu testen um einen NRF24L01+ Board defekt aus zu schließen?

knickohr, 2023-03-23:
Zeig mir mal das NRF-Modul von der anderen Seite.

AAANNNDDDIII, 2023-03-23:
Und nocheinmal prüfen, ob die SNR des Wechselrichters wirklich mit 1041... beginnt.

Joh49, 2023-03-23:
SNr 1041xxx sind die Mi600 von Hoymiles und nicht die HM... Da geht diese Lösung nicht

beegee3, 2023-03-23:
> SNr 1041xxx sind die Mi600 von Hoymiles und nicht die HM... Da geht diese Lösung nicht

bzw. HM600 Gen. 2 die das MI Protokoll benötigen. Das ist teilweise im Development Build enthalten, d.h. noch nicht alles funktioniert. Ist vielleicht einen Versuch Wert.

lumapu, 2023-03-23:
sehr gutes Auge @beegee3 😃

pade1984, 2023-03-24:
Danke euch! Gar nicht genau darauf geachtet. Dachte ich hätte ein HM-Wechselrichter.

Mit welcher DevBuild funktioniert es mit den MI-Wechselrichtern, dann probiere ich die mal aus?

beegee3, 2023-03-24:
das MI Protokoll wurde in den Dev. Versionen immer wieder verbessert, nimm daher die aktuelle Version: https://ahoydtu.de/web_install/ rechts: 'latest Development Build'.
zip Datei herunterladen und entpacken. Im Ordner 'firmware' findest du je nach ESP verschiedene .bin Dateien.
Im Ahoy DTU Menü 'Update' die richtige Datei auswählen (bei dir ..._esp8266.bin) und Update Button klicken.
Nach dem Update bootet der ESP neu. Danach sollte unten rechts bei GIT SHA ... die Dev. Version angezeigt werden (aktuell  0.5.105). Viel Erfolg!

pade1984, 2023-03-24:
Mit der Version 0.5.105 funktioniert es. Danke :-)

pade1984, 2023-03-25:
Ein Punkt ist mir noch aufgefallen: Mit der 0.5.105 ist keine Leistungsbegrenzung möglich. Arbeit daran schon jemand, damit das auch möglich wäre?

Hintergrund ist, dass ich einen LiFePO Speicher dranhängen möchte und darüber dann die Einspeiseleistung anpassen möchte.

beegee3, 2023-03-25:
wie gesagt, bei den MI Invertern läuft noch nicht alles rund. @rejoe2 arbeitet dran.
Bei der Leistungsbegrenzung gab es mit neusten Version 0.5.106 eine kleine Korrektur. Vielleicht geht's damit.
Bitte probiere nur das relative Limit (sprich %) aus, das absolute (W) wird m.E. noch nicht richtig übermittelt!

pade1984, 2023-03-25:
Ich habe auf die 0.5.106 aktualisiert, aber auch damit wird das relative Limit nicht wirklich umgesetzt. Also in den Logs steht, dass das Power Limit akzeptiert wurde, aber ich merke keinen Unterschied bei der Leistung.

@rejoe2 Hast du dazu Details?

Joh49, 2023-03-25:
zumindest bei mir ging die Leistungsbegrenzung mit der original Hoymiles DTU Pro auch nicht. Die Hoymiles Hotline hat mich an mehrere schlaue Leute verwiesen, welche jedoch auch keine Lösung hatten. Erst mit einem HM1500 Inverter ging es. Die Befehlübergabe wurde bestätigt jedoch nichts geschah.

rejoe2, 2023-03-25:
> zumindest bei mir ging die Leistungsbegrenzung mit der original Hoymiles DTU Pro auch nicht. Die Hoymiles Hotline hat mich an mehrere schlaue Leute verwiesen, welche jedoch auch keine Lösung hatten. Erst mit einem HM1500 Inverter ging es. Die Befehlübergabe wurde bestätigt jedoch nichts geschah.

Dann brauche ich mir also keine allzu großen Gedanken mehr dazu machen; ich hatte das auch schon so vermutet, weil mein MI-600 auch weder auf ZiyatT's Code noch auf den aktuellen dev-Versionen mit mehr als "hab's bekommen" reagiert (selbst, wenn an der richtigen Stelle die richtige Angabe kommt, was seit 106 der Fall sein sollte). Das gilt im Übrigen wohl auch für das "on"/"off"-Thema.
Die Limitierung mal nur mit CRC8 zu versenden hatte ich auch versucht; auch da (es ging mAn. korrekt raus): Keine effektive Umsetzung auf dem MI-600.  
Der einzige Test, den man noch hätte machen können, wäre eine absolute Limitierung gewesen, auch hier wieder mit CRC16 und CRC8. Aber dann schenke ich mir das wohl und warte weiter auf jemanden mit einem 4ch-MI...

closed 2023-03-27 by lumapu

Krantak, 2023-05-11:
Hallo Zusammen,

ich habe ein ähnliches Problem. Ich habe den HM-1500 mit Seriennummer 1161xxxxx.
Ich versuche seit einiger Zeit die DTU ans laufen zu bekommen, leider ohne Erfolg.
Ich bin hier den Anweisungen gefolgt und hatte als erstes auf die 0.6.9 stable geupdated.
Danach habe ich die Pins D3 auf D1 und D4 auf D2 umgebaut und in der Software umgestellt.
Danach bin ich auf die 0.6.12 hoch gegangen um zu schauen ob die Version vielleicht funktioniert.
Leider ist das Problem immernoch das gleiche. Keine Antwort vom Wechselrichter.
[image]
[image]
Das nrf24l01+ wird als verbunden erkannt. Wie gesagt es kommt halt nur keine Antwort vom WR. 

Hat hier vielleicht noch jemand eine Idee was falsch laufen könnte?

Danke und Gruß

Heiko

mtet2, 2023-05-11:
Hallo Heiko,

Der Wechselrichter Blinkt rot oder grün?
(PV-Module sind angeschlossen und produzieren auch)

Unter "settings Inverter" hast du definitiv die richtige SNR eingetragen.
Der Hacken bei "Communication Enable" ist gesetzt.
Die Intervalzeit > 20 s eingestellt
Die Zusatz-Elkos sind in der 3.3V und 5V Versorgung eingelötet.

Besteht eine Möglichkeit die "nrf24l01+" Platine gegen eine andere zu tauschen?

VG Manfred

knickohr, 2023-05-11:
Kontrolliert ob in den Settings - System Config die beiden LEDs auf off stehen ?

HorstyS, 2023-05-11:
> Kontrolliert ob in den Settings - System Config die beiden LEDs auf off stehen ?

Nur aus Interesse: wieso müssen die auf Off stehen?

Krantak, 2023-05-11:
Hallo Zusammen,

erstmal Danke für die schnelle Rückmeldung. 

@mtet2 Die Module arbeiten einwandfrei. Derzeit ernte ich ca. 150 Watt laut meinem Shelly. 
Seriennummer hatte ich als erstes Kontrolliert. Die passt. Communikation ist auch aktiviert.
Intervallzeit ist bei 30s. Ich habe derzeit einen 100my 16v Elko an dem RNF24. Wo soll denn der Elko in die 5V Versorgung? Die läuft bei mir derzeit über das Micro USB kabel des D1,

Ich habe heute nochmal ein Modul mit Antenne bestellt, da ich eh noch eine DTU für meinen Bruder bauen will. Werde dann damit mal einen Tausch durch führen oder die 2te DTU parallel testen.

@knickohr  Ja beide LED stehen auf off/default.

Ich habe nun auch noch mal alle Leitungen geprüft. Alles wie im Schaubild auf der Seite angeklemmt. Er scheint ja auch zu senden, es kommt nur keine Antwort. Werde die DTU Sa. noch mal mit zu meinem Bruder schleppen und da an seinem HM-1500 testen. Es sollte recht unwahrscheinlich sein, dass die Empfangseinheit bei beiden defekt ist.

Schönen Abend noch

Gruß

Heiko

rejoe2, 2023-05-11:
> > Kontrolliert ob in den Settings - System Config die beiden LEDs auf off stehen ?
> 
> Nur aus Interesse: wieso müssen die auf Off stehen?

"off" ist etwas zu weitgehend, wir haben nur leider das Problem, dass die zu oft auf falsche/störende GPIO verweisen. Daher (zumindest erst mal) ausschalten...

Wo kommt die Empfehlung mit den 30s Intervall her? Nach meinem bisherigen Eindruck (auch was Empfehlungen an andere, allerdings hauptsächlich MI-User angeht), ist es besser, das Intervall kürzer (aber nicht zu kurz) zu wählen. 5 Sek. scheint ein "guter" Wert zu sein.

Was Verkabelung angeht: Wenn "connected" angezeigt wird, ist das gut, verhindert aber weder, dass man Wackler/schlechte Lötverbindungen übersieht, noch zeigt es an, ob der IRQ-PIN sauber verdrahtet (und verlötet) ist. Dringende Empfehlung: Besser dreimal checken und messen! Zwischen dem jeweiligen PIN oben am ESP und dem nRF (ebenfalls oben am Modul gemessen) darf kein Widerstand messbar sein.     

Ansonsten ist es nicht toll, das in einem "closed" issue zu diskutieren...

ClausGra, 2023-07-23:
> > > Kontrolliert ob in den Settings - System Config die beiden LEDs auf off stehen ?
> > 
> > 
> > Nur aus Interesse: wieso müssen die auf Off stehen?
> 
> "off" ist etwas zu weitgehend, wir haben nur leider das Problem, dass die zu oft auf falsche/störende GPIO verweisen. Daher (zumindest erst mal) ausschalten...

Herzlichen Dank für diesen Hinweis. Er hat bei mir ein identisches Fehlerbild behoben!