lumapu/ahoy #799: REST API changes breaks integration - please document resp. keep API stable

GallusMax, 2023-03-24:
Zur Integration in andere Systeme ist eine REST API schön. Wenn die sich ändert, wirds schwierig.. 
Mir ist an den aktuellen Leistungswerten gelegen.

Mit dem aktiellen Release 0.5.66  war die Integration in ein Victron System recht einfach, weil andere Projekte die Werte unter /live auswerten .

Jetzt kommt der Bruch: in der Development Version (hier: 0.5.104) finden sich unter /live keine "live" Werte mehr. Scheinbar sind die Leistungsdaten nach /record/live umgezogen.
Selbstvertändlich sind Strukturänderungen manchmal sinnvoll, JSON Arrays sind nicht immer der Hit.

Dennoch meine Bitte:
- Beibehalten der einmal gewählten Schnittstellen - zumindest innerhalb einer Versionsreihe (von 0.5.66 nach 0.5 .xx?)
- Neue Strukturen dokumentieren und 
- idealerweise zunächst parallel unter neuer URL anbieten, vorhandene für den Releasewechsel abkündigen

Frage: was war die Motivation für die Array Struktur ? Parsern wird der Zugang so nicht leichter, oder?   
{
        "fld": "U_DC",
        "unit": "V",
        "val": "30.10"
      },

## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
https://github.com/lumapu/ahoy/commits/25695fc
Version:  0.5.104 (dev)
Github Hash: 25695fc

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [x ] Platform IO

### Desktop OS:
* [ x] Linux
* [ ] Windows
* [ ] Mac OS

dtuuser, 2023-03-24:
Hallo GallusMax,
vielleicht ist #737 hilfreich
Gruß

lumapu, 2023-03-25:
> Beibehalten der einmal gewählten Schnittstellen - zumindest innerhalb einer Versionsreihe (von 0.5.66 nach 0.5 .xx?)
> Neue Strukturen dokumentieren und
> idealerweise zunächst parallel unter neuer URL anbieten, vorhandene für den Releasewechsel abkündigen

würde ich sehr gerne machen. Man beachte aber auch, dass wir hier sehr stark begrent an Ressourcen sind und daher auch die Änderungen rühren. Bei der Dokumentation können wir uns noch verbessern.

closed 2023-03-27 by lumapu

dsteinkopf, 2023-03-28:
Ist die neue JSON-Struktur irgendwo dokumentiert? Ggf. freut mich auch ein Link auf eine hilfreiche Quellcode-Stelle. Danke sehr!

lumapu, 2023-03-28:
Dokumentation hinkt leider etwas hinterher, hier der Quellcode

https://github.com/lumapu/ahoy/blob/main/src/web/RestApi.h