lumapu/ahoy #800: [ESP8266] Reboot fails after firmware change from dev to release

GallusMax, 2023-03-24:
### Platform

ESP8266

### Model name

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.104

### Github Hash

 25695fc

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Linux

### Setup

Its about booting into another firmware only.

- Reboot device after successful save: [x]
- SAVE

### Error description

Running DTU Version: 0.5.104  (development branch with improved stability, but broken API)
Upgrade was flawless from 0.5.66 with 230319_ahoy_0.5.104_25695fc_esp8266.bin

Because of the broken/changed API I planned a downgrade to latest Release 0.5.66

1) Go to http://ahoy-dtu/update
2) upload 221230_ahoy_0.5.66_esp8266_f8fe044.bin
3) Reboot
4) and down we went 8-)
5) I expected the DTU to reappear after firmware change

Argafal, 2023-03-24:
If I was you I'd try to erase the flash, and then flash 0.5.66 from scratch. Then re-configure. 

The settings have changed in later dev versions, so I don't think you can downgrade from 0.5.104 to 0.5.66 without starting over. 

Curious to hear what you mean that the API is broken. Maybe you can open a separate issue for that? Fixing that surely must be a better solution than downgrading to 0.5.66, given that something like 0.5.105 will soon become the next release. ;)

GallusMax, 2023-03-24:
> If I was you I'd try to erase the flash, and then flash 0.5.66 from scratch. Then re-configure.
> 
> The settings have changed in later dev versions, so I don't think you can downgrade from 0.5.104 to 0.5.66 without starting over.

Thanks for the hint, flashing from  scratch is left over as last resort.
BUT: uploading a firmware and rebooting into it should always work. 
Ok - maybe it worked, but dropped the configured Wifi, showing up with an access point? At least its nasty from abroad. ;-)

closed 2023-03-27 by lumapu