lumapu/ahoy #801: Problem connecting to Inverter, RX success zero

veteranski, 2023-03-24:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ____nodemcu v3__
Retailer URL: ______az-delivery.de

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [x] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _0._5._66_
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [x] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [x] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

veteranski, 2023-03-24:
nrf24l01+
is connected
Datarate
250 kbps
Power Level
LOW
RX success
0
RX fail
0
RX no answer
9
RX frames received
0
TX count
49

veteranski, 2023-03-24:
[image]

veteranski, 2023-03-24:
12:43:57 I: enqueued cmd failed/timeout
12:43:57 I: (#0) I: no Payload received! (retransmits: 5)
12:43:57 I: resetPayload: id: 0
12:43:57 I: (#0) Requesting Inv SN 112183211696
12:43:57 I: (#0) sendTimePacket
12:43:57 I: sendTimePacket 5
12:43:57 I: TX 27B Ch3 | 15 83 21 16 96 80 65 41 16 80 05 00 64 1d 8c fd 00 00 00 00 00 00 00 00 64 2b 47 
12:43:58 W: while retrieving data: last frame missing: Request Retransmit
12:43:58 I: (#0) sendTimePacket
12:43:58 I: sendTimePacket 5
12:43:58 I: TX 27B Ch23 | 15 83 21 16 96 80 65 41 16 80 05 00 64 1d 8c fd 00 00 00 00 00 00 00 00 64 2b 47 
12:44:00 W: while retrieving data: last frame missing: Request Retransmit
12:44:00 I: (#0) sendTimePacket
12:44:00 I: sendTimePacket 5
12:44:00 I: TX 27B Ch40 | 15 83 21 16 96 80 65 41 16 80 05 00 64 1d 8c fd 00 00 00 00 00 00 00 00 64 2b 47 
12:44:02 W: while retrieving data: last frame missing: Request Retransmit
12:44:02 I: (#0) sendTimePacket

mtet2, 2023-03-24:
Ein Kondensator ca. 100 µF zwischen  +3,3 V und GND Eingang am "nrf24l01+ Modul" wirkt Wunder.
Auf die Polung achten!

veteranski, 2023-03-25:
ein 100 µF-Elko bringt keinen Erfolg. Zudem habe ich diverse GPIOS getauscht ohne Erfolg. Habe auch eine stärkere Versorgung mit 2A eingesetzt. Wie gesagt handelt es sich um eine v3 new nodemcu.
Was kann ich noch tun?

lumapu, 2023-03-26:
mal die aktuelle Dev-Version installieren: 
https://nightly.link/lumapu/ahoy/workflows/compile_development/development03/ahoydtu_dev.zip

Bitte mal das NRF auf Lötfehler prüfen, hatte schon eines mit einer Überlötung und funktionierte deshalb nicht.

air101, 2023-03-26:
Ich habe dieselben Probleme wie @veteranski. Habe bereits ESP8266 und ESP32 ausprobiert und die Firmware von @lumapu 
100 uF Kondensator bringt auch keinen Erfolg.

nrf24l01+
is connected
Datarate
250 kbps
Power Level
MAX
TX count
198
RX success
0
RX fail
0
RX no answer
197
RX fragments
0
TX retransmits
0

veteranski, 2023-03-26:
Habe ein 2. Set exakt gleich aufgebaut und siehe da, es funktioniert perfekt. Also entweder ist beim 1. Set der NRF oder die nodeMCU defekt.

closed 2023-03-27 by lumapu