lumapu/ahoy #802: absolutes Power Limit wird relativ gesetzt

gimehrzdbo, 2023-03-24:
Hallo,

(grosses Lob für das Projekt. Ahoy ist der Grund warum ich überhaupt ein Balkonkraftwerk mit HM WR gekauft habe.)

Ich habe den HM 1500 im Einsatz und verwende ihn mit 2 Paneelen mit je 410 WP
Diese sind beide nach Süden (gleiche Winkel nebeneinander) ausgerichtet und an Eingang 1 und 2 angeschlossen.

Nun will ich das persistente Limit auf 600 Watt setzen, damit ich hier systemkonform bin.
Im Ergebnis wird im Wechselrichter das Limit auf 40% gesetzt. 
Leider wird die Energie dann auch reduziert - meiner Erfahrung nach nicht ganz um 60% (eher ein paar Watt bis 1/3) - der "Fehler" wird möglicherweise immer größer, je höher die Ausgangsleistung des WR ist. (Verifizierung durch Variation der Werte als non persistent, absolut und auch relativ, gesetzt.)
Liegt das Problem im WR?
Mein Anliegen wäre, dass der WR unterhalb von 600W soviel produziert wie er kann und nur oberhalb der gesetzten absoluten Grenze, also über 600 Watt kappt.
Was ist zu tun?

## Hardware
  ESP8266

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module

### Power Stabilization:  Ohne

### Connection diagram: standard

## Software
AhoyDTU

### Version / Git SHA: 
Version: 0.5.66
Github Hash: _______

### Desktop OS:
Windows

dieewossmarthome, 2023-03-24:
Hallo, genau das gleiche Problem habe ich mit meinen 4 Module, nach der Begrenzung auf 40%, bringen auch die Module nur jeweils 40%. Auch wenn keine 600 Watt Ausgangsleistung erbracht wird.

Ollipop030, 2023-03-24:
Das liegt an den Wechselrichtern selbst. Der HM1500 hat zwei Tracker, auf denen die Leistung aufgeteilt wird. Wenn du das Limit auf 600W setzt, bekommt jeder Tracker 300W Leistung zugewiesen. Wenn jetzt zwei der vier Module im Schatten liegen, kommen von den Modulen mal angenommen noch 100W, von den Modulen in der Sonne dann die 300W. Sind zusammen 400 Watt Leistung, obwohl das Limit auf 600 steht.

gimehrzdbo, 2023-03-25:
Keine Verschattung auf einem der beiden Paneele. Beide sind im exakt gleichem Winkel montiert. Ja, sie sind auf einem tracker und die Idee dahinter ist, demnächst noch ein Modul zu ergänzen, welches gänzlich Richtung Osten zeigt. 

Das Problem worauf ich hinaus will, ist, dass der wr (oder ahoy) aus der absoluten Eingabe ein relatives Ergebnis macht. Eingabe: 600 Watt -> Limit 40%.
Mit dem weiteren Effekt, dass der Output entsprechend reduziert wird, auch wenn die Energie weit unter 600 Watt ist. Ich wünsche mir jedoch immer max Output bis 600 Watt.

Mart-Muc, 2023-03-25:
Für jeden  Tracker wird das Limit gesetzt. (ich habe 3P am HM1500)  V 5.66
Wobei relativ oder absolut keinen Unterschied macht. 
________________
!Zu Beachte ist, das Limit auf der DC-Seite gesetzt wird (soweit meine Erfahrung).!!!
—————————
Bei der Benutzung von zwei Panels (Eingang 1 und 3 wg der MPPT-Optimierung) ist es easy:
Also 600/0.95 = 631W setzen.

Bei ungerader Anzahl kann ein Limit nicht eingehalten werden!!
Ich habe unter Einbeziehung eines 4. virtuellen Panels ein Limit von 
600/3*4/0.95 eingestellt.
Somit wird der erste MPPT (linke Seite, 2 Panels parallel) korrekt auf 2/3 von 600=400W limitiert, die rechte Seite hingegen mit dem einem Panel nicht auf 200 sondern eben auch 400.

Würde ich das Limit auf 631W DC wie bei zwei belassen, wird im Bereich  der Einzelleistung 150-200W /0.95 der Panels bereits gedrosselt und Leistung „verschenkt“, obwohl die 600W AC noch gar nicht erreicht sind::
Z.B. 3x 180W /0.95 ergeben 300W + 180W AC Output.

für eine korrekte Drosselung auf das gesetzliche Limit ist somit entweder ein Inverter mit 4 (HMS2000, aber da geht Ahoy noch? nicht) oder zwei Inverter  (wirtschaftlich unsinnig) nötig. Oder aber aktive Limitierung (->enhancement Ahoy V200?)

Somit habe ich 600/3*4/0.95 bei drei Panels eingestellt, was ein „langsameres“ Überschreiten des dt Limits bedeutet aber auf jeden Fall unter dem Europäischen Limit bleibt.

Nachtrag wg AC/DC:
(Wobei mir dann nicht klar ist, wie bei dem HM1500 dann 1500W AC -Limit = 100% sein können??)

liebe Grüße Mart

gimehrzdbo, 2023-03-25:
Mich würde auch interessieren, ob die Anzeige "live" immer den Faktor anzeigt - auch bei Euch? Und das, obwohl ich den Wert absolut eingetragen habe...
[image]

Mart-Muc, 2023-03-25:
Ja, immer.
das wird einfach mit der Leistung des Inverters umgerechnet, da universell für alle Typen.
Wie gesagt- da ist kein Unterschied.
Das angegebene Limit wird gleich auf die MPPTs umgelegt:

in meinem&deinem Fall HM1500
2 MPPTs: (links 2 Panels parallel geschaltet, rechts 2 Panels parallel)
Deine Einstellung: max 631W DC = 2x 315W DC = 42%  =output 2x300W AC

closed 2023-03-27 by lumapu