lumapu/ahoy #803: Read/Display Firmware Version

rmayergfx, 2023-03-24:
Hi, i can´t find any information how to Display or to read the Firmware Version from the Inverter.

rmayergfx, 2023-03-24:
Wäre schön wenn das implementiert wird. Habe es bei OpenDTU gesehen das es dort verfügbar ist.

mtet2, 2023-03-24:
So zum Beispiel:
http://ahoy-dtu/api/record/info
Oder die IP von deiner "ahoy-dtu" einsetzen.

rmayergfx, 2023-03-24:
Danke, wusste nicht das der Wert in der Klammer die Firmware des Inverters ist. Die Api gibt aber noch viel mehr aus. 
Leider habe ich das in der Doku nicht gefunden sonst hätte ich hier kein Issue geöffnet. 
Wäre halt schon schön wenn man das auch per mqtt bzw im Webinterface in einer eigenen Rubrik direkt sehen könnte.

Würde sich doch Prima anbieten im Bereich System einen Punkt Inverter mit in die Oberfläche zu integrieren und dort die Daten aus der Api zu hinterlegen, Firmwareversion mit Datum z.B.

lumapu, 2023-03-25:
per MQTT wird es bereits übertragen:

[image]

rmayergfx, 2023-03-25:
Ich habe es in HA (Docker Synology mit Eclipse Mosquitto) eingebunden. Dort hat er keine der von dir gezeigten Entitäten gefunden oder eingebunden, auch der MQTT Explorer http://mqtt-explorer.com/ zeigt keine dieser Werte an. Ist allerdings auch ein TSUN TSOL-M800

lumapu, 2023-03-26:
Es gibt zwei verschiedene MQTT Pfade, einmal das was du hier nennst und per `Autodiscover` angelegt wird und einmal die live Daten aus dem Wechselrichter. 
Unter den Live-Daten des Wechselrichters befindet sich auch die FW-Version

closed 2023-07-23 by lumapu