lumapu/ahoy #805: FW v0.5.106 bug second device with new name created

rmayergfx, 2023-03-25:
Nach dem Update auf FW v0.5.106 hat das System nun ein zusätzliches Gerät "AHOY_DTU_DF7C" angelegt das auf der gleichen IP liegt... 

Downgrade auf v0.5.105 und das zusätzliche Gerät verschwindet wieder.

Argafal, 2023-03-25:
Ich glaube das ist ein neues Feature in 0.5.106, wenn ich das richtig verstanden habe. Statt AHOY_DTU gibt es jetzt einen eindeutigen Namen. Für die die nur ein Device haben ist es egal, aber für die die mehr als einen Testaufbau haben hilft es die verschiedenen Geräte zu unterscheiden. Im Discord gibt's glaub ich ne Anleitung wie man das umbenennen kann, falls man es wieder nur als "AHOY-DTU" haben will.

dtuuser, 2023-03-25:
Bei mir ist der Name geblieben und es ist in der Fritzbox kein weiterer erschienen

Argafal, 2023-03-25:
Ich nahm an, dass @rmayergfx vom MQTT auto discovery spricht. Vielleicht hab ich es falsch verstanden. Warten wir mal bis er/sie erklärt von welchem "System" geredet wird. Ja, der hostname für den Router ist glaube ich gleich geblieben. 

Wäre das folgende ein guter Änderungsvorschlag? Man verwendet immer den eingestellten Hostname als Device name in MQTT. Der hostname kriegt als Default ein geräteabhängiges Suffix. Ein Benutzer der lieber einen anderen Hostname und Device name hat, kann das Gerät dann einfach selber in den ahoy Einstellungen umbenennen.

rmayergfx, 2023-03-25:
Also in den Ahoy Settings steht bei mir für MQTT nicht der default "inverter" sondern M800. Nun ist ein 2tes Device im MQTT aufgetaucht mit dem Namen "AHOY_DTU_DF7C" und dieses weitere Gerät hat 4 Sensoren. Ich habe aber nur eine AhoyDTU im Netzwerk und nun in HA ein weiteres MQTT Gerät das eigentlich gar nicht existieren dürfte, da es ja bereits das Gerät gibt.
Der Inverter hat im Setup den Namen TSUN_M800 und dort sind auch soweit alle Sensoren vorhanden.

Was mir hier jedoch fehlt sind nach wie vor die MQTT Informationen zu Firmware und HWPartId, diese sind nur per MQTT Explorer sichtbar, tauchen jedoch nirgendwo in HA auf.

lumapu, 2023-03-26:
Hier geht es um `Autodiscover`. Das zweite Gerät sind die Totalwerte, also Summen über alle Inverter. Evtl. sollten wir das publishen der Totalwerte nur bei Multi-Inverter-Setups machen.
Stört das Gerät wirklich?

rmayergfx, 2023-03-26:
Ja, da es nur zur Verwirrung führt.

beegee3, 2023-03-26:
Totalwerte liefern nur bei Multi-Inverter-Setups neues, machen so gesehen nur dort Sinn.
Das 'Total' Gerät vielleicht mit dem Device Name aus dem Setup + '_Total' benennen, dann ist die Inverter Zuordnung klar.

rmayergfx, 2023-03-26:
Wofür ein "Total" Gerät bei einem Single Setup? Alle Werte dort wie gehabt einpflegen und für ein Multi-Inverter-Setup eine Option in der Firmware einpflegen das dort ein zustätzliches Gerät "Multi-Total" optional aktiviert werden kann. Dann ist es eindeutig und führt bei Single Setups nicht zu Verwirrungen.

beegee3, 2023-03-26:
> Wofür ein "Total" Gerät bei einem Single Setup? Alle Werte dort wie gehabt einpflegen und für ein Multi-Inverter-Setup eine Option in der Firmware einpflegen das dort ein zustätzliches Gerät [...]

'Total' nur bei Multi-Inverter, so meinte ich das! Dann aber sollte der 'Total' Gerätename nicht fix sein, sondern von der Ahoy-DTU abhängen, sonst gibt es Probleme, wenn jemand mehr als eine DTU einsetzt. Und da bietet sich der im Setup konfigurierbare Device Name an (mit angefügtem "_Total"). Ist leichter zu identifizieren, als das jetzige "AHOY_DTU_XXYY" mit XXYY aus der MAC erzeugt.

lumapu, 2023-03-26:
@beegee3 @rmayergfx : Umbau ist schon im Gange, bin gerade beim Test

closed 2023-03-27 by lumapu