lumapu/ahoy #806: ESP32_S3 proper pin assignment for SPI/IRQ/CE not possible

markusdd, 2023-03-25:
### Platform

ESP32

### Model name

ESP32_S3 openDTUFusion v1 (see Discord)

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

    -DHOYMILES_PIN_MISO=48
    -DHOYMILES_PIN_MOSI=35
    -DHOYMILES_PIN_SCLK=36
    -DHOYMILES_PIN_IRQ=47
    -DHOYMILES_PIN_CE=38
    -DHOYMILES_PIN_CS=37
    -DARDUINO_USB_MODE=1

(this is essentially the working build config from OpenDTU, with Ahoy this cannot be done this way)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

master

### Github Hash

ddfe33a666b276eba669524c46c374e73ef79b12

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Windows

### Setup

invalid, the correct config cannot be given through the GUI right now and has been hardcoded into the firmware, see below.

### Error description

This is about the discussion in DIscord roughly starting here: https://discord.com/channels/984173303147155506/1060347063281401876/1089216043412623492

In a quest to bring up Ahoy DTU for the openDTU Fusion v1 board that I developed and the community reviewed, which uses a ESP32-S3-WROOM-1U module, we disocovered that Ahoy DTU runs perfectly fine on the S3, but due to hardcoded pins in the code and some GPIOs missing from the GUI dropdowns, it is currently impossible to make it work without hacking the code.
This also might/will affect any other commercially available ESP32-S3 kits as well, shoudl they use some of the higher order GPIOs or some that were blocked as input only on the legacy ESP32.

Here are the changes I 'hacked to make it work.
First, within the ahoy repo I did this to enforce the correct pins for CE, CS and IRQ:

[image]
[image]
[image]

As, in contrast to openDTU, ahoy uses the RF24 function to initialize the SPIClass the problem is that MISO, MOSI and SCK do not get exposed as a config option, although the SPI driver supports it no problem. 
So I also hacked the RF24.cpp to enforce the correct pin initialization. I might have overlooked where this gets handed in from ahoy, but it did the trick for now:
[image]

This is really the only thing that is needed. Make MISO, MOSI and SCK selectable in the GUI and add missing GPIOs to the GUI dropdowns so they become valid config options.

With the hacked firmware, AhoyDTU works perfectly well on the S3:

[image]

[image]

Here is also the relevant part of the schematic of the OpenDTU Fusion PCB the community is currently testing/developing:
[image]

knickohr, 2023-03-25:
Das betrifft jetzt aber erstmal die Kommunikstion. Wir werden wohl noch weitere solche „Hacks“ benötigen für die LEDs und das Display 😵

Aber erst mal das wesentliche 😉

markusdd, 2023-03-25:
Addition, above diffs were delivered on main.
I was just educated that development03 is where the action is.

Will try to see what needs to be changed there to make it work.

markusdd, 2023-03-25:
it appears the necessary change on the SPI side is to use anotehr constructor of RF24:
[image]

This one allows to also customize the spi bus pins

markusdd, 2023-03-26:
Die eigentlichen SPI pins MISO, MOSI, SCLK.

Die aktuelle implementierung nahm bisher an (was für den ESP82XX) auch richtig war, dass die immer gleich sind.
Für den ESP32 geht das für die meisten üblichen devkits auch ncoh gut, bei ESP32-S3/C3 geht das aber völlig in die Hose, weil dort kann der SPI einfach überall hingelegt werden.

Konkrete Motivation war die entwicklung vom openDTUFusion Board wo ein S3 drauf ist. Da kommt der SPI auf pins raus, die es ncihtmal in der auswahlliste gab.

Ist aber schon gelöst das Problem, siehe PR hier: https://github.com/lumapu/ahoy/pull/807

markusdd, 2023-03-26:
ich würde mal grob ebhaupten das problem ist gelöst udn funktioniert.
Wenn es keine gegenteiligen Meinungen gibt wegen mir hier close, es wurde ja laut Discord auf allen PLattformen udn mit allen Display getestet

lumapu, 2023-03-26:
oder ich close beim nächsten Release, denke das passiert die Tage

closed 2023-03-27 by lumapu