lumapu/ahoy #810: Problem connecting to Inverter, RX success zero | ESP8266 v4 USB-C

DcFamas, 2023-03-26:
### Platform

ESP8266

### Model name

Wemos Lolin D1 mini v4.0 ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

[image]

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

Just WIFI and Inverter.
[image]

### Debug Serial Log output

```shell
15:01:26 W: while retrieving data: last frame missing: Request Retransmit
15:01:26 I: TX 11B Ch75 | 15 81 81 43 13 83 68 94 32 83 8b 
15:01:27 I: resetPayload: id: 0
15:01:27 I: (#0) enqueuedCmd: 11
15:01:27 I: (#0) sendTimePacket
15:01:27 I: sendTimePacket b
15:01:27 I: TX 27B Ch3 | 15 81 81 43 13 83 68 94 32 80 0b 00 64 20 50 37 00 00 00 64 00 00 00 00 fd 7b 42 
16:01:28 W: while retrieving data: Frame 1 missing: Request Retransmit
16:01:28 I: TX 11B Ch23 | 15 81 81 43 13 83 68 94 32 81 89 
16:01:29 I: procPyld: cmd:  11
16:01:29 I: procPyld: txid: 0x95
16:01:29 I: resetPayload: id: 0
16:01:29 I: (#0) enqueuedCmd: 11
16:01:29 I: (#0) sendTimePacket
16:01:29 I: sendTimePacket b
16:01:29 I: TX 27B Ch40 | 15 81 81 43 13 83 68 94 32 80 0b 00 64 20 50 39 00 00 00 64 00 00 00 00 9d 37 60 
16:01:30 W: while retrieving data: last frame missing: Request Retransmit
16:01:30 I: TX 11B Ch61 | 15 81 81 43 13 83 68 94 32 83 8b 
16:01:31 I: resetPayload: id: 0
16:01:31 I: (#0) enqueuedCmd: 11
16:01:31 I: (#0) sendTimePacket
16:01:31 I: sendTimePacket b
16:01:31 I: TX 27B Ch75 | 15 81 81 43 13 83 68 94 32 80 0b 00 64 20 50 3b 00 00 00 64 00 00 00 00 fd 2e 1b
```

### Error description

Inverter doesn't show up with my USB-C Wemos D1 Mini. I haven't changed anything and used the 100uF capacitor. Because I ordered a PCB I'm not able to change the wiring on the fly. What could I try?

lumapu, 2023-03-26:
wird das NRF erkannt? (wenn nicht ist eine Warnung auf der Index.html)
`your NRF24 module can't be reached, check the wiring and pinout`

In den Einstellungen kann ich davon ausgehen, dass die Pins `CE` und `CS` korrekt eingestellt wurden. Wie bei so vielen gerade rate ich dazu eine Optische Inspektion des NRF24 zu machen, ich hatte eins da war eine Überlötung und hat deshalb nicht funktioniert.

DcFamas, 2023-03-26:
Modul wird erkannt. Habe es mit verschiedenen ESP8266 und verschiedenen NRF24 getestet, sogar die mit der Antenne waren dabei. :/

lumapu, 2023-03-26:
versehe ich nicht, dann muss man sich die Verdrahtung nochmal im Detail anschauen. Nicht, dass einmal von oben und einmal von unten drauf geschaut wurde ...

Argafal, 2023-03-26:
Ich würde an deiner Stelle versuchen mal den IRQ von D3 auf D0 zu legen. Wenn das nicht hilft, dann IRQ zurück auf D3 und CE statt D4 auf D0 legen.  Musst du vielleicht ein bisschen tricksen, nachdem du das fest verdrahtet auf der Platine hast. Natürlich auch jeweils in den Einstellungen ändern.

D3/D4 haben in der Vergangenheit bei bestimmten D1 Boards für Probleme gesorgt, daher der Vorschlag.

DcFamas, 2023-03-27:
@lumapu hab schon alle Pins durchgepiepst, sind strikt nach Anleitung verdrahtet.
@Argafal danke für den Hinweis, werde ich später probieren und mich zurückmelden. Ist dann natürlich echt doof. :/

Loetnase, 2023-03-27:
Hast du eventuell so einen D1 Mini noch zur Hand, dieser Typ hat bei mir immer problemlos nach Schaltplan funktioniert.
Vieleicht ist deiner so ein Problemtyp welcher hier zum Untersuchen noch gesucht wird. 
[image]

closed 2023-07-23 by lumapu