lumapu/ahoy #812: Warning, do not update to dev release 0.5.108 or greater if you don´t have the pinout on your desk

rmayergfx, 2023-03-27:
Dev. Fix 0.5.108

    merge: PR SPI pins configureable (ESP32) #807, #806

Will break your connection to your nrf24l01+ as the additional GPIO Settings will be set to default.
in my case with a board from Solarmodule-Gladbeck the nrf24l01+ is no longer available until you change
the default settings. My working settings was for this board is:
CS: GPIO5
CE: GPIO33
IRG: GPIO27

both LED off, but red one still is on.

markusdd, 2023-03-27:
as long as they do not have some exotic ESP32 board yes. (as ahoy could never change these pins that should hold for all users, only OpenDTU could do that until now).

Argafal, 2023-03-27:
So far no one could have had an ESP32 board with Ahoy that wasn't 19/23/18. So I don't expect that to be an issue. Maybe @rmayergfx can change the title of this issue, cause one doesn't need a specific pinout to resolve this, it's always the same. It just needs to be manually set once.

markusdd, 2023-03-27:
let me make a quick PR that updates the docs as well.

rmayergfx, 2023-03-27:
So when it is not changed why does a user need to setup this after update? why not set this to default settings instead of 0 after update?

markusdd, 2023-03-27:
because we have ne way to tell. If there is a 0 in the flash at that position, that could well mean pin0.

Only if we would see an erased cell at 255, that could prompt us to use the default.

This si s developing project, and the non-configurability of SPI pins was a shortcoming coming from the 8266-only days.

There can be instances where an update introduces slightly breaking changes. Also, this is a devlopment branch. You should be able to deal with something like this, or not use it.

rmayergfx, 2023-03-27:
What about a hint in the GUI near the Settings to show the "default" Settings? Just to give a hint what GPIO should be there?
e.g.  SCLK  (def. GPIO18)
Think this will be great to new or non technic users.

markusdd, 2023-03-27:
The defaults only apply to standard ESP32 boards.
For new ESP32-C3 and S3 there is absolutely no way to tell what a sane default is.
And for new users which flash a new device the defaults will be set.

This only happens to users that have a pre-existing config.

Please test by making a full erase and flash it like a new device.
Then the pin setting should be the correct default.

rmayergfx, 2023-03-31:
Siehe https://github.com/lumapu/ahoy/issues/838
Da werden auch noch mehr kommen. Können wir nicht per Default die Standardwerte setzen?

duracell, 2023-04-01:
Ah, ich bin auch in die Falle getappt :(
Könnte man es ansonsten nicht in die Release-Notes schreiben? (Also wenn es nicht irgendwo stand und ich es übersehen habe.)

markusdd, 2023-04-01:
es steht in den release notes des commits, und auch im getting started. leider nicht in denen der release area. @lumapu kann man das ggf noch fixen?

lumapu, 2023-04-01:
@markusdd ich hoffe so wie ich es jetzt eingetragen habe ist es korrekt

markusdd, 2023-04-01:
> @markusdd ich hoffe so wie ich es jetzt eingetragen habe ist es korrekt

Traumhaft! Danke dir.

closed 2023-04-19 by lumapu