lumapu/ahoy #817: 0.5.66 Live View stops producing values if MQTT connected

jensgol, 2023-03-27:
I have an ESP8266 Wemos Mini D1 setup running fine with 0.5.66 for 2 weeks in my setup.

Now I tried to use MQTT (using mosquitto). So Ahoy was found, config is transmitted and looks good.
MQTT Explorer shows all settings and channels. Good.

BUT:
I get only 1 data package to MQTT and the Ahoy "Live" View stops producing values every 30s.
After some minutes the "connected" field shows "0".

I can reproduce the effect, when I remove the MQTT IP from Ahoy settings, so Live View runs fine again, standalone.

When I enter the IP from the MQTT Broker the Live view shows 1 data package after reboot (MQTT too) and then stops and show these values forever.

Seems to be an bug?

rmayergfx, 2023-03-27:
Check first your connection on Serial/Connection tab. Is there any issue in the communication with the nrf24l01+ ?
Do you have to much retransmits or fragments? This can also be checked on System tab.
How is the hardware setup? Do you have an capacitor applied for the nrf24l01+ ?

jensgol, 2023-03-27:
Please delete my issue, apparently the cat has crashed my DTU from the window this morning, so the antenna had fallen down ;-) Works fine now ;-)

rmayergfx, 2023-03-27:
So please ask in the future first your cat about any issue :-)

closed 2023-03-27 by lumapu