lumapu/ahoy #818: Please add ESP32 WROOM 30Pin Layout to readme

rmayergfx, 2023-03-27:
Already a solution here: https://github.com/lumapu/ahoy/issues/683
Hard to find for new users. Please add to redme/docu

markusdd, 2023-03-27:
This has been done in this PR already.
https://github.com/lumapu/ahoy/pull/814/files

rmayergfx, 2023-03-27:
@Argafal 
No your guess is total wrong. Give someone who has found the information about AhoyDTU or OpenDTU via google and wants to setup his own device. He buys an ESP32 and there are different boards online. 38Pin, 30Pin. So when you go on the Readme.md, you will find only an esp8266 and ESP32 with 38Pin, but if you get e.g. an ESP32-Wroom recommended by some people, they will stuck on the wiring and are confused because of only 30Pin and different wiring. 
believe me, i was asked about the wiring from a newbie with not much experience and he was confused about the setup.

markusdd, 2023-03-27:
For this OpenDTU provides device profile files.

Ahoy has no such mechanism (yet), but in it's documentation provides guidance to buy very SPECIFIC boards.
If you decide to deviate from that, you are on your own right now.
If you feel so strongly about supporting documentation for another board, there is a good opportunity to draw the pictures and submit a PR, just like I did to support SPI selection at all and to enhance the docs about this change.

This is a community effort and no one is employed for this. Your statement is valid, there are no docs for the 30pin modules. So: Please provide this documentation. (I cannot do it, I do not own one of these)

rmayergfx, 2023-03-27:
The documentation for 30Pin is already done  in #683 just needs to copy this to documentiation to easier find this.

btw. i can´t find a recommandation which board to buy! 
In Getting_started.md'thins-needed everyone find "ESP32 Dev Board NodeMCU WROOM32 WiFi", finding with your favourite search engine you will end on different sites and see most of them have only 30Pins. 
Inside the wiki i can´t find any information.  The informations are available across all platforms, but it is hard to find them. Some are in Readme.md, some in Getting_Started.md, a few inside Wiki, also on Discord are good informations. It would be helpful to have a singel point where the basics are collected.

How long do you think was my search when i started with AhoyDTU to find the information about wiring a display and why it ist not working with the default web setup.  in the deep of the web i found with a search engine the entry to install a dev. Version to get the display running. Why not posting the information on the homepage that the stable version that will install via web installer does not support a display out of the box?

I like to help and i am writing about things every newbie is fighting with.

Argafal, 2023-03-27:
Feels like we all need to take a deep breath. :) I want to try to suggest a constructive way forward. 

I agree that the documentation can be improved. The information isn't always very clear and sometimes a bit scattered. I can appreciate that it's not so easy for someone who's just starting.

@rmayergfx: I wonder, would you be able to do the following: make a copy of any webpage content, readme file, documentation text, picture that you feel should be improved. And can you change them? And then you post them here, or you make a pull request out of the files. I'll volunteer to help you get these changes integrated. But someone needs to work out the exact new text and where it goes. Maybe your recent (admittedly frustrating) experience can be really useful for that. :)

rmayergfx, 2023-03-27:
Would be very helpful to clearify first of all together with @lumpau and other people working on this to get a useful and good solution where to post or insert the missing data. Should we expand the wiki in the Hardware Section and put a link on all other Readme and Getting startet files just to keep them small and collect all important data inside the wiki ? Or is it necessary to put some information into the get started? 

I think it should be splitted into 2 different main topics. 
on the info side there should be the information what kind of inverters are supported. This list can be upgraded, i think there will not be too much changes in the future.

Hardware should explain the ESP8266 and ESP32 Mainboards, with a sub for the nrf24l01+ and important warnings about fakes and how to identify them (just as goodie)

Software should inform about stable and nightliys and what is working an not and this should be the main topic with all explanations what is going on and planned.

Argafal, 2023-03-27:
From my side: sounds good, please go ahead as you suggested! :+1: 

As to question about the where: You were the one who said that the information wasn't where you were looking for. So I think you should make a suggestion how to change the places where you actually looked (and didn't find what you needed).

stefan123t, 2023-12-21:
This is similar to #762 please double check whether this is already addressed there ?