lumapu/ahoy #823: [ESP8266] MQTT topic inverter/MC3/ch0/# publishing ceases minutes after reboot

GallusMax, 2023-03-28:
### Platform

ESP8266

### Model name

ESP8266 NodeMCU

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D1 |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.105

### Github Hash

5a1a600

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0,1,2
   
    - Name: HM-1200
    - Active Power Limit: none
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375 x 3 x 4
### General
    - Interval [s]: 20
    - Max retries per Payload: 4
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: (hidden, Victron venus broker)
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D2
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: HIGH
#### Serial Console
    - print inverter data: []
    - Serial Debug: []
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

I have configured three Hoymiles HM-1200 with 4 DC inputs each named MC1,MC2,MC3.

After reboot the inverters show up and deliver P_DC and P_AC values. The values appear in the web view /live fine and for all three inverters.

If looking at the mqtt topics, only MC1 and MC2, (index 0 and 1) work ok and stable.
Inverter MC3 also reports for a short time after rebooting, especially in the inverter/MC3/ch0 topic. Publishing then stops ind the ch0 topic, which is the only place where P_AC is reported.

0) reboot Ahoy, and keep an eye on inverter topics. mqtt explorer is your friend
1) Watch topic inverter/MC3/ch0/P_AC or inverter/MC3/ch0/P_DC
2) about one minute after rebooting the topic stalls, while other topics receive updates (screenshots below)

Regarding mqtt config "mqtt fixed interval". 
I tested both settings: 
a) setting an interval of 0 - updates on demand
b) setting an interval of 20s - updates at fixed times
This setting seems of no relevance: on both cases MC3 (inverter #2) gets no more publishing.

The screenshots show the relevant topics. MC1 and MC2 get updates, but MC3 hangs. 
N.B: This is while MC3/ch1/P_DC still gets updates!
 
[image]
[image]
[image]

Argafal, 2023-03-28:
Probier mal mit 0.6.0 :) Da gab es einen MQTT fix.

GallusMax, 2023-03-28:
Danke! Mach ich, sobald ich mich traue (von den neuen defaultpins habe ich
gehört :-)

Noch ne Frage: ist es möglich, MISO und SCL (fix auf D5 und D6)
konfigurierbar zu machen? Es geht um ein Board mit festmontiertem Display,
das I2C auf D5 und D6 nutzt. Da fällt die Wahl ganz schön schwer: entweder
Display oder nRF24 ;-)

viele Grüße
Ulrich

Argafal, 2023-03-28:
MISO und SCLK des NRF-Moduls? Hat doch @markusdd gerade einstellbar gemacht, ist in 0.6.0 drin.

GallusMax, 2023-03-31:
> das geht beim ESP8266 aber nicht. Die 'alte Möhre' kann das nicht. Wenn da also i2C und SPI wirklich übereinander liegen sieht es schlecht aus. Evtl. mal gucken ob man via SOftware den I2C umlegen kann?

(sorry for off-topic) Kann man sicher umlegen, aber, da das Display hart verdrahtet ist D5/D6 :-)  Danke trotzdem für den Tip für die alte Möhre!

GallusMax, 2023-04-03:
> Probier mal mit 0.6.0 :) Da gab es einen MQTT fix.

Upgrade to v0.6.0 - leider weiterhin: MQTT publishing von ch0/P_AC erfolgt eher sporadisch. Bisweilen liegen Minuten zwischen den regelmäßigen publishes der ersten beiden inverter und dem dritten. (die Screenshots oben haben weiter Gültigkeit)

Es sieht so aus, als ob der "letzte" angeschlossene Inverter / der mit dem höchsten Index, und von dem der ch4 und ch0  betroffen ist. 
(Ein Posting von einer Anlage mit zwei Invertern nennt auch den mit dem Index 1 - siehe 
posting zu #833 )

These: Sowohl die Werte ch4/P_DC als auch die ch0/P_AC des jeweils letzten Inverters fallen beim mqtt publish sporadisch aus. Die Werte unter /live oder /api/inverter/id/n sind dagegen auf Stand bzw. brauchbar.

(Könnte das an den begrenzten Ressourcen auf dem verwendeten Board 8266 liegen? )

lumapu, 2023-04-05:
ich glaube es liegt an der Lib. Wir müssen hier die Systematik ändern, mit der die Daten gepublished werden. Aktuell wird eine Schleife über alle Inverter gemacht und das sehr schnell. Hierdurch hat die MqTT Lib nicht genügen Zeit und auch nicht genügend Speicher die Messages zwischenzuspeichern. 
Mir kommt das so bekannt vor, da ich gleiches auch beim Autodiscover hatte. Das Phänomen auch hier war, dass immer am Ende was gefehlt hat. Die Lösung war letzendlich die Schleife über die Haupt-loop `void loop() {...}` laufen zu lassen und keine Interne `for-loop` zu haben.

lumapu, 2023-07-23:
das Problem liegt an der MqTT lib. Diese hat eine verkette Liste, die im Speicher dynamisch erzeugt und wieder gelöscht wird. Das klappt super auf PCs, nicht aber auf Mikrocontrolern. Ich werde in Kürze für die MqTT-Lib einen PR einreichen.
In Ahoy wurde der oben angekündigte Ansatz zwischenzeitlich umgesetzt und behebt genau dieses Problem, da die Liste somit nur aus einem Element besteht und der Speicher nicht fragmentiert wird.

closed 2023-07-23 by lumapu