lumapu/ahoy #825: Device information MQTT "von Hoymiles"

rmayergfx, 2023-03-28:
Where does HA get the information "from Hoymiles" via MQTT? I have a TSUN inverter here and it is a bit irritating.
Is this fixed in the firmware?

rejoe2, 2023-03-28:
Don't really understand the question, but if your TSUN is a 2nd gen. Device (serian nr. 10x1), recent AhoyDTU firmware (0.6.0) should be able to request the informations this type provides. Pls. note: 2nd gen. Devices provide less information than newer models.

Apart from that: Please provide more info.

rmayergfx, 2023-03-28:
What do you need on information. Synology Docker with installed acutal HA and Eclipse Mosquito also as Docker. Nothing else.Enabled MQTT in HA and AhoyDTU, device was recognized and when you click on the MQTT Devices you will see all devices detected by the system. There is the MQTT name given in AhoyDT with underlined information "from Hoymiles". This can´t be changed, so this information was deployed by AhoyDTU i think. I never did write anywhere Hoymiles, because i have a TSUN.

rejoe2, 2023-03-28:
OK, so you are just taking about the info displayed within HomeAssistant. 
Can't help further, I'm using FHEM as automation sofrware, but I'd recommend to rename your "issue" to make your point more clear.

beegee3, 2023-03-28:
yes, the manufacturer ("mf") in the JSON sent to HA via MQTT is hardcoded as "Hoymiles".

beegee3, 2023-03-28:
as far as I know HA needs the "mf" information. Since most users have Hoymiles inverters it's hardcoded.
I don't know if there's much interest in changing that. But this is a question for @lumapu. It may be that the manufacturer can be deduced from the serial number.

lumapu, 2023-03-28:
using the serial number will be the approach I would choose.
This feature has no priority for me, if you need it on a short term don't hesitate to publish a pull request

closed 2023-07-23 by lumapu