lumapu/ahoy #826: Prevent sending not updated data over MQTT

rejoe2, 2023-03-29:
## Hardware
  * [ ] ESP8266
  * [ x] ESP32
  * [ ] Raspberry Pi

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version:0.6.0

If AhoyDTU querries more than one inverter, each time one inverter is updated, a lot of (all production related?) datapoints from all inverters are sent on the MQTT side (see below for one sample datapoint).

Would it be possible to restrict the send data to the updated inverter and the consolidated DTU data?
This would lower the entire traffic and prevent (potentially) outdated data to be regarded as up to date.

Or are there other mechanisms implemented to especially prevent processing outdated data at all? 
 
```
07:18:08.205	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	7.8
07:18:11.939	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	7.8
07:18:16.173	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	7.8
07:18:24.431	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.2
07:18:28.205	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.2
07:18:36.892	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.2
07:18:41.177	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.2
07:18:46.176	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.2
07:18:53.219	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.2
07:18:58.280	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.6
07:19:02.106	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.6
07:19:11.215	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.6
07:19:18.176	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.6
07:19:22.165	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.6
07:19:27.960	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	8.6
07:19:34.189	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9
07:19:38.840	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9
07:19:46.189	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9
07:19:53.028	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9
07:19:58.272	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9
07:20:04.193	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9
07:20:10.220	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.4
07:20:23.167	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.4
07:20:27.958	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.4
07:20:34.224	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.4
07:20:38.823	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.4
07:20:44.197	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:20:47.955	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:20:56.266	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:04.196	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:08.111	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:15.403	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:21.207	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:31.702	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:39.191	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:44.232	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:49.560	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	9.6
07:21:55.028	AHOY_DTU_F5A184	inverter/O_Sued/ch0/P_DC	10.2
```

knickohr, 2023-03-29:
Ja, das ist mir auch schon aufgestoßen. Hat mich aber nur tangiert und ich konnte damit leben. Jetzt mit 5 Invertern wird das schon lästig.

Das ist auch die Ursache warum immer wieder mal Nullwerte oder alte Werte für einen Inverter übertragen werden obwohl der Betroffene Inverter noch schläft.

beegee3, 2023-03-31:
im wesentlichen werden nur **alle** `RealTimeRunData_Debug` Daten immer wieder übertragen, sobald **ein** Inverter neue Daten hat. Um das erneute publish von "alten" Daten zu vermeiden hier ein Vorschlag, wo beim RTR publish der aktuelle Timestamp des Inverters gespeichert wird. Es wird nur noch übertragen, wenn gespeicherter und aktueller Timestamp verschieden sind
(@lumapu reicht RTR, oder sollte man das auch für die anderen Infos nutzen?):

in `pubMqtt.h` definiere: `uint32_t mIvLastRTRpub[MAX_NUM_INVERTERS];`
in `PubMqtt()` initialisieren: `memset(mIvLastRTRpub, 0, MAX_NUM_INVERTERS * 4);` (* 4, da uint32_t = 4 bytes!)
zuletzt `sendData` anpassen:
```
        void sendData(Inverter<> *iv, uint8_t curInfoCmd) {
            record_t<> *rec = iv->getRecordStruct(curInfoCmd);

            uint32_t lastTs = iv->getLastTs(rec);
            bool pubData = (lastTs > 0);
            if (curInfoCmd == RealTimeRunData_Debug)
                pubData &= (lastTs != mIvLastRTRpub[iv->id]);

            if (pubData) {
                mIvLastRTRpub[iv->id] = lastTs;
                for (uint8_t i = 0; i < rec->length; i++) {
                    ...
                }
            }
        }
```

rejoe2, 2023-03-31:
Habe das mal in meine dev03 eingebaut. Gefühlt gibt es immer noch zu viele updates, aber weniger wie vorher, und ob mIvLastRTRpub[] an die richtige Stelle gewandert ist, kann ich nicht sicher sagen, hatte anders ein paar komische Fehler beim compilen (already defined)...

PS: Das timing-Limit für die micros sind jetzt mal auf 410 gesetzt. Bei meinen früheren Tests hatte ich irgendwann ein faible für ca. 200ms-Zeitscheiben entwickelt, für meine Inverter ist das ok, und der spontane Eindruck war, dass das auch für die 3rd gen. ruhiger aussieht.

beegee3, 2023-03-31:
die Definition gehört ans Ende von pubMqtt, z.B. direkt unter die von mLastIvState

rejoe2, 2023-03-31:
Gut. Da hatte ich sie in der Tat auch hingesetzt; mein Bauchgefühl scheint demnach ok zu sein, was sowas angeht *grins*.

closed 2023-04-19 by lumapu