lumapu/ahoy #830: Support ESP32 C3

Argafal, 2023-03-29:
I am trying out a ESP32-C3. The board is a Lolin C3 mini, v2.1.0. In this issue I will collect changes required to run Ahoy on an ESP32-c3. If ever successful (big if) this can be used for a pull request later. Contributions/tests of others are most welcome.

My build environment (add this to platformio.ini):
```
[env:lolin_c3_mini]
platform = espressif32
board = lolin_c3_mini
build_flags = -D RELEASE -std=gnu++14 -DCONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=true -DCONFIG_ESP_CONSOLE_UART=ESP_CONSOLE_USB_SERIAL_JTAG
build_unflags = -std=gnu++11
monitor_filters =
    ;default   ; Remove typical terminal control codes from input
    ;time      ; Add timestamp with milliseconds for each new line
    ;log2file  ; Log data to a file “platformio-device-monitor-*.log” located in the current working directory
    esp32_exception_decoder
```

Argafal, 2023-03-29:
First problem: a boot loop. The error looks as such:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x25b0
entry 0x403cc710
I: Initializing FS ..
I:  .. done
[   205][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
W: failed to load json, using default config
I: Settings valid: false
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
```

It is caused by non-existing pins being selected in the default configuration. Pin 16 (irq) does not exist on the C3 board. Hence in hmRadio.h the setup(...) method fails when calling `pinMode(irq, INPUT_PULLUP);`

The user should be informed when this fails. One can work around this issue by changing the default pins in config/config.h. Ahoy then boots.

Argafal, 2023-03-29:
Next up, no access point is created. Ahoy keeps on trying to connect to the wifi with the ssid `YOUR_WIFI_SSID`. 

I changed config/config.h to include my ssid and wifi password. However, AHOY doesn't connect to it either. Log snippet:
```
scanning APs with SSID Cinderella
reconnect in 8 seconds
reconnect in 7 seconds
reconnect in 6 seconds
reconnect in 5 seconds
reconnect in 4 seconds
reconnect in 3 seconds
reconnect in 2 seconds
Wifi event: 1
reconnect in 1 seconds
BSSID 0: 0 11 32 9d 20 95
reconnect in 0 seconds
try to connect to AP with BSSID: 0 11 32 9d 20 95
reconnect in 19 seconds
Wifi event: 5
I: connectionEvent
reconnect in 18 seconds
Wifi event: 5
I: connectionEvent
reconnect in 17 seconds
Wifi event: 5
I: connectionEvent
reconnect in 16 seconds
Wifi event: 5
I: connectionEvent
reconnect in 15 seconds
Wifi event: 5
I: connectionEvent
reconnect in 14 seconds
Wifi event: 5
I: connectionEvent
reconnect in 13 seconds
Wifi event: 5
I: connectionEvent
reconnect in 12 seconds
Wifi event: 5
```

I have not found a solution yet. I will update this comment if/when I do. Others might chime in. Note: this is v2.1.0 of the Lolin C3 mini board. v1.0.0 had a wifi hardware issue. This version does not (supposedly).

stefan123t, 2024-10-27:
@lumapu are there any plans to support the ESP32-C3 too ? Or can we close this as `wontfix` ?

stefan123t, 2025-09-27:
@bentolor Benjamin, the ESP32-C3 is a different architecture all together, ie. RISC-V based whereas ESP32-S3 is based on the legacy Espressif Xtensa LX7 Dual-Core processor architecture.

The issue you reported looks very much like the pin out does not match the config of your build, i.e. it is pretty much the same error that @Argafal reported some time ago.

There were discussions to stop supporting the ESP8266 in AhoyDTU going forward past version 0.8.156 and OpenDTU-OnBattery has already moved on to only support ESP32-S3 N16R8 with 16MB Flash memory lately.
So you may want to make a wise decision on which platform / chip to solder into your project, depending on how smooth the experience should be. 
For ESP32-C3 I would expect some serious efforts by yourself to port the AhoyDTU code to the RISC-V architecture and polish some of the rough edges of this truly preliminary work-in-progress.

So in principle it should be possible to get it to work on ESP32-C3 there are probably still a lot of unknown factors and uncharted code ahead. Good luck and always a stack trace under your debugging code!

bentolor, 2025-09-28:
Hi Stefan @stefan123t, thank you for your detailed and valuable feedback. 

Honestly, I wasn't really aware even of this platform mixup in the ESP32 space. I only used [this ESP32 comparison (german article with a table)](https://cool-web.de/esp8266-esp32/espressif-esp32-chip-varianten-im-vergleich.htm) listing this huuge ecosystem of different, and felt there must be newer & better options than the WROOM-32 from nearly a decade ago. **I assumed all of them fall under the ESP32 label**.

For the ahoyDTU: I think it's wise to focus on known working platforms & chips. I don't see value in putting beyond trivial effort on new chips. With that in mind I'm still little puzzled about the ESP-C3 binary in the release files. I assume this is for a different board.

My suggestion: at some place of the `Getting_Started.md` actually **list** the supported Models (and Boards). As listed above. I could prepare a MR for the section, but I don't have the knowledge to feed it's content.

I also feel it's beyond time to deprecate ESP8266: On all sorts of ESP use cases their limitations cause troubles all over the places in my limited experiences. Given that WROOM-32 only has 4 MB flash, I wonder what OpenDTU-OnBattery does with 16MB flash: Do they ship DOOM?

### 2 Questions:

1. So regarding my options: Beyond WROOM-32: What options are recommend for Ahoy? i.e., I do have an  ESP32-S3 N16R8 in stock. Could I use that without any road bumps ahead?
2. Shall I prepare a MR for the docs?

stefan123t, 2025-09-28:
@bentolor we develop for the OpenDTU-Fusion board which is based on the ESP32-S3 models. They also provide now N16R8 chips on board. The reason for the 16 MB Flash is that OpenDTU-OnBattery as well as AhoyDTU v0.9.xxx will provide additional functions which have to be stored in code somewhere. As OTA Flashing is the hot thing for most of the users, we can only use about 50% of flash mem for the code, which brings us down to less than 2 MB for the 4 MB models only.

While it may be time for ESP8266 there are still a lot of users choosing AhoyDTU as it also supports the "old" ESP8266 models. Though there are efforts under way to port the OpenDTU/AhoyDTU Hoymiles library to the ESPHome platform, which would allow to user ESP8266 without the overhead of the AhoyDTU user interface.

Regarding the ESP32-C3 there seem to have been some efforts in the two Issues you mentioned, though calling them supported is premature. So for the time being I would simply keep the Getting Started as is. Do you see any need to change what is listed there ?

https://ahoydtu.de/getting_started/ has ESP-D1 mini and ESP32 WROOM listed as examples. There will probably be a change once v0.9.xxx comes out. (It has been in development secretly for over a year now, afaik)