lumapu/ahoy #832: [ESP8266] No data from inverter due to corrupted LED pin settings

tastendruecker123, 2023-03-29:
### Platform

ESP8266

### Model name

NodeMCU

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
not relevant
```

### Error description

I haven't been able to reproduce it reliably yet, but I have run into this problem several times over the past days.

Ahoy works as expected if I input all the settings in one go and save them. When making small changes to the settings several times, at some point the values for the LED pins were changed from '255' to '0'. It's possible that part of the settings page hadn't loaded properly (see issue #828).

Because '0' refers to pin D0, and D0 is the default IRQ pin, Ahoy was using the IRQ pin as a LED output at the same time, which means it was unable to receive any inverter data.

I did not change the LED pins manually and I never touched any of the pin settings at all because my setup uses the default wiring scheme. I suspect that Ahoy defaults to '0' when there's no valid data submitted for the LED pins, which breaks the NRF pin configurations.

I would suggest making sure that the two LED pin settings always default to 255, and perhaps add a check to make sure that the LED pins do not conflict with the NRF pins.

I'm mainly submitting this bug in case someone else runs into it as well. It's no big deal if you know about it and you can simply change the LED pins, but it's quite tricky to find if you don't and the inverter just stops communicating all of a sudden.

Joh49, 2023-03-29:
the workaround I performed at version 0.6.0 was to change in the setup - display config: 
the pin configuration clock and data to off (default) 
and it worked as before

Joh49, 2023-03-29:
this may be only applicable at the 1.3" SH1106 display

closed 2023-04-19 by lumapu