lumapu/ahoy #833: [ESP8266] mqtt topics at inverter/total missing?

GallusMax, 2023-03-29:
### Platform

ESP8266

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

n/a

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

n/a

### Error description

1) Subscribe to inverter/total/YieldDay oder inverter/total/P_AC
2) no values published any more
3) I expected a value for daily yield and current sum of AC Power published as in previous release / version 0.5.66

GallusMax, 2023-03-31:
This issue could be a result of #823 - the inverter/total topics are said to be filled only when all inverters are "visible". 
As described in #823 , the third inverter (topics named here:  inverter/MC3/ch0) get no publishes , and because of aged values there the inverter/total stalls, too.
N.B.: the third inverter must be "visible", as the inverter/MC3/ch1/P_DC gets regular publications (publishes? :-)
Hint: this is of version 0.5.105 - I will update as soon I can test 0.6.x

GallusMax, 2023-03-31:
Update: mqtt topic inverter/total/P_AC (for example) gets no updates, even shortly after DTU restart, and three inverters answering ok.
(still 0.5.105, though)

Manos1966, 2023-04-02:
My first comment in GitHub, please excuse me if it is not helpful:
Currently using two inverters (HM1200 and HM1200Garten)
As of Ver0.6.0, 
MQTT sends messages for the First Inverter.
MQTT sends ONLY following messages for the second inverter:
MQTT2_DEVICE PV_Inverter P_DC1_HM1200Garten: 0.8
MQTT2_DEVICE PV_Inverter P_DC2_HM1200Garten: 17.2
MQTT2_DEVICE PV_Inverter P_DC3_HM1200Garten: 0.3
MQTT2_DEVICE PV_Inverter P_DC4_HM1200Garten: 20.2
No other information related to the second inverter is transmitted.

Both Inverters online and producing:
Board: ESP32
Uptime: 0 Days, 00:55:33
ESP-Time: 2.4.2023, 12:46:48
System Infos:
Inverter #0: HM1200 (v10016) is available and is producing
Inverter #1: HM1200Garten (v10012) is available and is producing
MQTT is connected, 21832 packets sent, 0 packets received
You are using the current stable release: 0.6.0

GallusMax, 2023-04-03:
> My first comment in GitHub, please excuse me if it is not helpful: 
Indeed: this was very helpful and supports in short, what I filed a few days ago.

In my case its the 3rd inverter (also the last one in the array) , that offers incomplete data. More specific: the P_DC are updated regularly, but the P_AC are published "less often" than for the first two inverters.  (as described in detail in  
issue #823 )

I switched to 0.6.0 meanwhile, but inverter/total/P_AC does not get publishes, (perhaps still because of too great  differences of timestamps between the P_AC values? just guessing?)

I have a screenshot of the only time total/P_AC had a value. The value was 0 and it turned up once the sun was down and all three inverters delivered P_AC=0 ;--) The timestamps of the ch0/P_AC values differ about 4min, 20sec.

System Infos:
Inverter #0: MC1 (v10018) is available but no data was received until now
Inverter #1: MC2 (v10018) is available but no data was received until now
Inverter #2: MC3 (v10018) is available but no data was received until now
MQTT is connected, 54 packets sent, 0 packets received
Polling inverter(s), will pause at sunset 3.4.2023, 20:57:00
You are using the current stable release: 0.6.0

Manos1966, 2023-04-04:
Hi again,
I found something that maybe of help, although I myself can not understand completely.
It looks to me as if AHOY MQTT sends data under two different names
Here is one example:
My original "**AHOY_DTU:inverter**/total/P_AC:.* P_AC_total" delivers nothing
Instead, "**AHOY_DTU_FD6CD4**:inverter/total/P_AC:.* P_AC_total" now delivers the right values

GallusMax, 2023-04-04:
> Hi again, I found something that maybe of help, although I myself can not understand completely. It looks to me as if AHOY MQTT sends data under a new name, including part of the MAC address Here is one example: My original "**AHOY_DTU:inverter**/total/P_AC:.* P_AC_total" delivers nothing Instead, "**AHOY_DTU_FD6CD4**:inverter/total/P_AC:.* P_AC_total" now delivers the right values

Thanks for your input - please help me understand: in both cases you mentioned the topic "inverter/total/P_AC" (which does not deliver in my installation, too) . What is the part it in front of the colon? Do you use different MQTT servers (aka broker)?

lumapu, 2023-04-04:
geht es hier jetzt um reines MqTT oder um Autodiscover von Homeassistant?

Manos1966, 2023-04-04:
I use FHEM
I presume it is the autodiscover of FHEM.
The previous AHOY versions have identified themselves with "AHOY_DTU" ("inverter" I entered myself in the MQTT of AHOY).
Now I noticed, with the new version, the autodiscover of FHEM identifies an "AHOY_DTU_FD6CD4" (FD6CD4 is part of the MAC of my ESP32). I believe I found all data under the "AHOY_DTU_FD6CD4" messages.

I currently have a salad, some data come as AHOY_DTU and other as AHOY_DTU_FD6CD4, but have no time to look, I am out of the house.

I hope it helps a little
Manos

closed 2023-07-23 by lumapu