lumapu/ahoy #834: [ESP8266] prometheus /metrics endpont wrong encoding

mlausch, 2023-03-29:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

I use victoria metrics to scrape the /metrics endpoint, but I get the following error in the log
`cannot scrape target "http://192.168.68.66:80/metrics" ({instance="192.168.68.66:80",job="ahoy"}) 1 out of 1 times during -promscrape.suppressScrapeErrorsDelay=0s; the last error: error when scraping "http://192.168.68.66:80/metrics": unexpected char ' ' at the end of chunk size. Expected '\r'`

The HTTP transfer is chunk encoded (-> transfer-encoding: chunked). The HTTP spec says, that after every chunksize and data-chunk a '\r\n' must be send.
From the ahoyDTU I get only ' \n'. The victoria metrics parser seems to be very unhappy with that.

I have just added a haproxy setup between victoria and the ahoyDTU which sanitize this successfully. But I think the issue should be fixed in the ahoyDTU code itself.

Thanks,
Manuel

fsck-block, 2023-03-31:
Hi Manuel

Had a look at this issue

> From the ahoyDTU I get only ' \n'. The victoria metrics parser seems to be very unhappy with that.
> "http://192.168.68.66:80/metrics": unexpected char ' ' at the end of chunk size. Expected '\r'

As far as i can see on network level it's not a problem of the single '\n' for each metric. 

As the error message of victoria metrics shows it a problem of the transmitted chunk size.
RFC 2616 chapter 3.6.1 (Chunked Transfer Coding) defines the chunk size as `chunk-size     = 1*HEX` following `CR LF`
Unfortunately the used ESP Async WebServer adds spaces to the end of the chunk size to get a four character length chunk size.
ESP Async WebServer/src/WebResponses.cpp line 313
```
    if(_chunked){
      // HTTP 1.1 allows leading zeros in chunk length. Or spaces may be added.
      // See RFC2616 sections 2, 3.6.1.
      readLen = _fillBufferAndProcessTemplates(buf+headLen+6, outLen - 8);
...
      outLen = sprintf_P((char*)buf+headLen, PSTR("%x"), readLen) + headLen;
      while(outLen < headLen + 4) buf[outLen++] = ' ';
```
In the opinion of ESP Async WebServer it seems to be legal to add spaces so it may be difficult to fix this on short-term

fsck-block, 2023-04-01:
Hallo Manuel

Falls du ahoy selbst baust, gibt es Möglichkeiten zu schnellen (lokalen) fix.
Du könntest enen fork für das Repository  "yubox-node-org/ESPAsyncWebServer" erstellen und dort einen kleine Änderung  vornehmen. Dann nur noch in deinem lokalen platformio.ini file auf deinen Fork zeigen lassen und ahoy bauen.

Die Änderung ist eher trivial. Sorgt dafür dass die Chunk-länge immer größer als 4 Zeichen ist. Dann wird nicht mit spaces aufgefüllt:
```
$ diff ESP\ Async\ WebServer/src/WebResponses.cpp*
321c321
<       outLen = sprintf_P((char*)buf+headLen, PSTR("%04x"), readLen) + headLen; // https://github.com/lumapu/ahoy/issues/834
---
>       outLen = sprintf_P((char*)buf+headLen, PSTR("%x"), readLen) + headLen;

```

closed 2023-08-17 by lumapu