lumapu/ahoy #840: REST API view broken on chrome based browsers -- Firefox 111.0.1 still working

rmayergfx, 2023-03-31:
REST API view on chrome based browsers is broken. only textlines are viewed. This happens on actual versions of Chrome and Edge.

Firefox view still working.

[image]

lumapu, 2023-04-01:
This must be a change from the browsers side. We didn't changed anything. The basic REST API URL `/api` reports a valid JSON.

rmayergfx, 2023-04-05:
Just a short update. If anyone uses Chrome and need a good solution to view the REST API agin, install from the chrome webstore the addon JSON Formatter from callumlocke.com
https://chrome.google.com/webstore/detail/json-formatter/bcjindcccaagfpapjjmafapmmgkkhgoa?

closed 2023-08-05 by lumapu